robincornelius/libedssharp #13: After adding a new SDO, it is not accessable in PDO mapping dropdown

s-fuchs, 2016-11-03:
PDO-Map flag was set to optional. 
Current workaround: save the XML and open it again..

robincornelius, 2016-11-03:
Hmm seems to be working here. If i create a new object, as soon as i change the PDO type to optional and hit save they are available in the PDO drop downs and the list of available PDOs.

The save button should fire the doUpdatePDOs() which should propagate through and force the PDOs to completely refresh what is available.

robincornelius, 2016-11-04:
Looks like there are a few different things going on here. 
* Adding a Var object and setting PDO map to optional works.
* Adding a Rec object, then adding a subindex and setting its PDO mapping to optional also works
* Adding an array object and setting the pdo map to optional does not work with out saving and reloading. So the Parent PDO mapping is not propagating through the sub objects.

I've also spotted when checking this morning that the parent entry for a REC object should grey out the PDO options but currently it allows this to be set on the parent when only the sub objects make sense.

So the fault is array objects.  This leads to another problem we must consider. Although you can do as your patch added and allow the pdo field to be set on all subobjects it presents a confusing situation when exporting to canopennode.  The XML and EDS files both support fine grained permissions for PDO for subobjects so this is no problem. But when generating the main object dictionary in CO_OD.c arrays only have one flags field for the entire array so it is impossible to pick and choose different fields. (This is not the case for REC objects, these have individual field flags).

I would suggest the following solution:-

1. Allow array subobjects to have individual PDO mappings and access types set, is is consistent with can open EDS specs and supported in EDS and XML saves.

2. When exporting to canopennode only use the parent property for the subobjects, if any subobjects differ in PDO/access permissions to the parent flag a warning message to the user

3. Also in the GUI when setting the PDO/access permissions on the parent, automatically propagate this to all subobjects so this provides a quick way to set the entire array but also allows individual array elements to be set independently if required

s-fuchs, 2016-11-04:
You are right, i did not think this all the way through. 
Your solution sounds good to me. But i don't think it is worth the effort if the canopennode code does not support this granularity. Just your third point would be enough (and omitting my patch).

robincornelius, 2016-11-06:
I've implemented a patch that updates the subobjects and fixes the basic bug described here but not gone as far as allowing the individual ones to be set without the parent

commit a95411fa (2016-11-06): Propagate PDO parameters for child objects of arrays when updating them in GUI. Fixes #13
closed 2016-11-06 by robincornelius