robincornelius/libedssharp #29: Change attribute of "predefined error field[0]" in generated CO_OD.c from 0x86 to 0x8E

m1ke02, 2016-11-21:
otherwise the field cannot be cleared:

{0x1003, 0x10, 0x86, 4, (void*)&CO_OD_RAM.preDefinedErrorField[0]}
needs to be changed to
{0x1003, 0x10, 0x8E, 4, (void*)&CO_OD_RAM.preDefinedErrorField[0]}

robincornelius, 2016-11-21:
In the object dictionary editor page change the Access type from ro to rw hit save, then export the CO_OD.c again. It looks like the default CanOpenNode IO_Example.xml has r/o permissions as default on object 0x1003. I'll update the xml i ship as the default example to have r/w permission as it makes more sense

m1ke02, 2016-11-21:
The problem is whole PEF array must be read-only, except for the "Number of errors" field. As it seems to me, this requirement of CANOpen standard cannot be met with any default OD settings and must be hard-coded as an exception.

robincornelius, 2016-11-21:
Ah yes i understand. Changing the access type to r/w will generate the correct code because CanOpenNode does not look at subobject access permissions for arrays. It only sets the flags based on the entire array. The way it enforces correct access is to use CO_OD_configure() to override default behavior then the correct permissions are enforced in c code in the function CO_ODF_1003() not in the access to the OD as generated by this tool. 

So at the moment in my EDSEdtior just setting the entire object as access type r/w will do the correct thing. I have added a new access type to the latest code that can be used to reset the correct behavior which may be helpful if using the tool to make EDS files. Both the EDS and XML file formats do support independent permissions for child objects.

commit 2b0915f3 (2016-11-21): 0x1003 permission updates, fixes #29
* Add new access type to GUI 0x1003 rw/ro to reflect the special permissions of object 0x1003. This type is not directly saved but will apply
rw permissions to subobject 0 (and the parent) and ro permissions to the rest.
* Update default profile to have correct permissions for 0x1003 and subindex of 0 (regression)
closed 2016-11-21 by robincornelius