robincornelius/libedssharp #38: CO_DATA.c/h Export

jimlittlefield, 2016-12-07:
Using version 0.4 it looks like the export to CANopenNode is not quite right.    The c/h files compile ok and the node sort of runs although it appears to stuck in the pre-operational state.       There is also an immediate Protocol Emergency Message on bootup.

Attempting to access COB 0x1800 I'm able to read subindex 0,1,2,3 but attempting to read 4,5,6 always returns a "Sub Index Does not exist" error ... even though the return value from a read of subindex 0 tells me the highest supported index is 6.

I'm reasonably confident that my CANopenNode code is working correctly because I'm able to successfully do all these things using CO_DATA.c/h generated by another tool.

Thanks
Jim

ps - Overall...like the UI of the tool much more than the web based OD editor that comes w/ the CANopenNode release!

m1ke02, 2016-12-07:
I'm using 0.4 too. But with my xml states can be changed freely, no emergency message on bootup, [0x1800;0x05] (event timer) reads also OK.

[Drive.xml.txt](https://github.com/robincornelius/libedssharp/files/637695/Drive.xml.txt)

robincornelius, 2016-12-07:
Hi Jim,

Sounds like something has gone wrong with the PDO mapping/parameters creation. The error code you are getting on the bus i expect is :-

CO_EMC_PROTOCOL_ERROR           0x8200

with bit

CO_EM_PDO_WRONG_MAPPING 0x1A 

This error is critical and a device may not move to operational state.

Any chance you can send some files through, either attach to this report or email myself directly so I can take a look. There are a couple of possibilities of things that may have gone wrong including 0.4 does not support dummy PDO maps (0.5/git does) There were also some issues spotted where some device eds files do not have complete indexes and skip entries (especially compatibility entry in the PDO communication parameters) I have partially fixed this but there could still be bugs here

It also depends how you started your file? was it using file ->new then importing the profile? or did you load an existing canopennode project.xml or did you import an eds?

If you can let me know what steps you took and also ideally send me the files then I'll fix this.

robincornelius, 2016-12-07:
Its probably also worth checking the PDO mapping tabs show sensible stuff and also directly look in the object dictionary editor page and see what the contents of 0x1800 and 0x1a00 is

jimlittlefield, 2016-12-07:
Hi Robin,

Thanks for the quick response.    I started with an EDS file which I had 
generated from the Web OD Editor that comes w/ CANopenNode.   
Unfortunately I can only intermittently get that tool to startup.    At 
this point I've managed to thrash around enough to break pretty much 
everything so I'll backtrack,   start with a know good 
c/h/_project.xml/eds file set and then apply your tool and see what 
happens.    If I encounter the same issues I'll attach all the relevant 
files to my earlier comment.

Regards,

Jim

robincornelius, 2016-12-07:
Ok cool,

I'm still happy to take a look at your original eds if you like, its actually good testing feeding in files such as you describe to see why the final output broke.

If you want to start fresh either use New in my tool then select Insert DS301 Profile, this will give you a generic canopen profile with enough to start a project. Or load in the _project.xml from the CanOpenNode git repository that you can find in the example/ folder https://github.com/CANopenNode/CANopenNode/blob/master/example/_project.xml this is my reference test and it should produce the same CO_OD.c/h as in the example/ folder (actually there is an issue where this file is out of sync with what was committed but that is just some trace entries and fairly obvious)

jimlittlefield, 2016-12-08:
Thanks for the info.    I'll try your xml file to see if I get better results.

Jim

jimlittlefield, 2016-12-08:
Per Robin's suggestion,  I started with the base CANopenNode  _project.xml file instead of the eds.    Generating an initial c/h file set from this worked OK!    I've been able to do some editing of the default settings and the c/h files still work so I think most of my problems stem from not starting with a good EDS file.

Robin....thanks for offering to look over my eds.    I think you got me started in the right direction and your tool is doing useful things.    I'll post a few other wrinkles I've encountered as part of another thread.

Regards,
Jim

jimlittlefield, 2016-12-08:
As this appears to be related to the eds file rather than the edssharp tool I'm closing the issue.

closed 2016-12-08 by jimlittlefield