robincornelius/libedssharp #39: Device Settings on Device Info Tab....

jimlittlefield, 2016-12-08:
I've tried checking/unchecking the various options under "Device Settings' and also changed the "Granularity" parameter,   Update, and then export c/h.     Version control tells me that there is no change to the exported c/h files.      I was thinking these might map to the  macros in CO_OD.h which define the FEATURES ....
/*******************************************************************************
   FEATURES
*******************************************************************************/
  #define CO_NO_SYNC                     1   //Associated objects: 1005-1007
  #define CO_NO_EMERGENCY                1   //Associated objects: 1014, 1015
  #define CO_NO_SDO_SERVER               1   //Associated objects: 1200-127F
  #define CO_NO_SDO_CLIENT               0   //Associated objects: 1280-12FF
  #define CO_NO_RPDO                     21   //Associated objects: 14xx, 16xx
  #define CO_NO_TPDO                     4   //Associated objects: 18xx, 1Axx
  #define CO_NO_NMT_MASTER               0

Also curious about what the "Concrete Node Id" setting does?

Thanks ...jim

s-fuchs, 2016-12-08:
Most of these options don't affect CANopenNode, so they are only present in the eds and not in c/h-files.

"Concrete Node Id" can be used with the "Report" feature "Network PDO". If you open multiple xml with corresponding node IDs, the report shows you how the PDOs are connected.

jimlittlefield, 2016-12-08:
Thanks for the info!    I guess this is not a bug but perhaps me expecting something different.   I looked on the wiki and also in what I got from git-master.    Is there a documentation file/faq/manpage

closed 2016-12-08 by jimlittlefield

robincornelius, 2016-12-08:
The main problem is i've not got around to writing any kind of documentation.

When running reports the concrete node id will replace the $NODEID dummy variable, which is part of the CanOpen EDS standard. Quite often the PDO COB-ID will be expressed as $NODEID+0x200 for example so that when you set the actual node id this will be replaced and a real value used for the report