robincornelius/libedssharp #40: Number of RPDO's

jimlittlefield, 2016-12-08:
I added 17 additional RPDOs using the Rx PDO Mapping tab (right click -> insert PDO), then verified that the additional RPDO communications/mapping objects were listed in the Object Dictionary.     Saved settings and returned to the Device Info Tab where the Number of RPDOs remained at 4 (initial value).    Exporting the c/h files does produce the desired 21 RPDOs however.

I then went back to the RX PDO Mapping tab and mapped an object to the 5th RPDO (first 4 are already defined).     After saving the change and going back to the Device Info screen the number of RPDOs is now correctly shown as 21.

I did this starting with the _project.xml released w/ CANopenNode.

jl

robincornelius, 2016-12-08:
Fixed in git, will be in next 0.5 release

closed 2016-12-08 by robincornelius

jimlittlefield, 2016-12-09:
Thanks Robin...you fix stuff pretty fast!

commit 211df33c (2016-12-08): * PDOs are not recounted when adding new, removing or inserting a new profile. Fixes #40

They are recounted if the file is saved and loaded again. If PDOs are manually added to
the object dictionary then they still will not be counted until the file is save/loaded

* Default DS301 profile contained an example PDO mapping to an object that is not included within the profile