robincornelius/libedssharp #41: Adding Sub-index to 0x1016 max-subindex increment error

jimlittlefield, 2016-12-10:
I just retrieved libedssharp sources and rebuilt (VS2013).    Using the build result the Object Dictionary Editor is v0.5.

1.   Load the DS301_profile.xml into editor
2.   Select Object Dictionary 0x1016 Consumer heartbeat time.   The max sub-index starts at 4 and there are 4 subindexs defined.
3.    Right click sub-index 0 and select "Add sub item".     The max-subindex increments by 1 and the new items appears in the list.   (all good!)
4.    Continue adding subitems until the max sub-index is at 10.
5.    Add one more sub-item :  The max sub-item index changes from 10 to 17 even though only one additional sub-item was added to the subindex list.   Perhaps the 10 displayed was interpreted somewhere as a hex value (16 + 1) = 17?
6.    If you keep on adding sub-items the max sub-item index will increment by larger amounts eventually exceeding the range of UNSIGNED8 and causing the editor to throw and exception.

The work around is to manually reset the max sub-item index value (Right click on sub-index 0 and select the option to change the max sub-index).      I've fooled around with this a bit and I'm pretty sure one part of the editor code interprets the max sub-index as a hex value but other parts assume its decimal.

jl

robincornelius, 2016-12-12:
Thanks, found the issue, it is a conversion issue as everything gets stored as strings internally because of the EDS/XML output formats and its also allowable to have any default value number in decimal, hex or octal. The max sub index value was not going through the correct converter. 

Ideally the max subindex needs to be extracted by a function within libedssharp not in the GUI but that is for another day..

closed 2016-12-12 by robincornelius

jimlittlefield, 2016-12-12:
Thanks Robin...I will rebuild using your latest.
jl

commit 9df069dc (2016-12-12): Use EDSSharp.ConvertTo functions instead of the default c# ones as mine are number base aware
and the default value that comes from an EDS may be in decimal,hex or octal as per the CanOpen EDS
specification. Fixes #41