robincornelius/libedssharp #45: RPDO default configuration disabled.

jimlittlefield, 2016-12-31:
I have some RPDOs that I want configured to start/default to the disabled state.    This would involve setting bit 31 of  subindex 01 for the RPDO communications OD entry.   

0x1404,1  COB-ID = 0x200 + $NODEID
 
Using the editor changed this to 0x80000200 + $NODEID hoping that the default configuration would have the associated RPDO to be disabled.     My application code would later clear bit 31 to enable the RPDO.    

After making this edit I was able to save the change, save the device xml file but on attempting to reopen the device xml file I receive an error...

Error parsing node id 0x3+0x80000200 nodes, System.OverflowException: Value was either too large or too small for a UInt16.
   at System.Convert.ToUInt16(String value, Int32 fromBase)
   at libEDSsharp.EDSsharp.GetNodeID(String input) in c:\Users\jal\Downloads\libedssharp-master\libedssharp-master\libEDSsharp\eds.cs:line 1433

Since COB-ID is UNSIGNED32 I presume that the editor is checking for overflow of the lower 16 bits of the value.    

How can I create RPDO/TPDOs which are disabled initially?

Thanks
Jim

robincornelius, 2017-01-03:
Looks like i'm parsing the data at some point into a Uint16 which is incorrect as the field is a UInt32.

The GetNodeID() function which has the job of parsing $NODEID only currently returns a UInt16, which usually makes sense as cobs are 16 bit, however as you found in some cases a 32 bit is allowed so i need to adjust this to cope with the full possible width that it may encounter

closed 2017-01-03 by robincornelius

jimlittlefield, 2017-01-03:
Hi Robin...thanks for the fix.   I like the checkbox for invalid bit....good convenience feature!

commit 5ac2e353 (2017-01-03): Use UInt32 for COBID as COBID != NODEID and has extra stuff in upper bits such as invalid flag fixes #45
Also add check box for setting invalid bit from PDO configuration pages