robincornelius/libedssharp #47: Licensing of generated files

martinwag, 2017-01-24:
Hi,

I'm basically having the same problem as with the original js generator ([https://sourceforge.net/p/canopennode/discussion/387151/thread/a56d0cc8/](url).

When I create the CANopenNode object dictionary files, they contain a LGPL license header. As there is no dynamic linking in deeply embedded devices, this means that those files in practise can't be used in commercial applications in that use case.
The files in the CANopenNode github repo contain a header with linking exception, e.g. [https://github.com/CANopenNode/CANopenNode/blob/master/CANopen.c](url). According to Janez Paternoster it's OK to change the generated files to the same license too (see first link).
Is it OK for you to change the license for your files also?

Thanks!

robincornelius, 2017-01-24:
Hi,

Short answer yes thats fine.

Longer answer. The generated file when used with the xml file from CanOpenNode produces exactly the same file (+/- errors) as in CanOpenNode, therefor i am not claiming any kind of ownership of that file and in fact i've left the copyright "Copyright (C) 2004-2008 Janez Paternoster" so as it is basically a copy of the file from CanOpenNode so the linking exceptions will apply in exactly the same way as with CanOpenNode itsself.

I probably should edit the generator to include this information and a link to the forum where this was discussed for clarity. So i'll leave this issue open to remind my self to update when i have time.

Thanks!

martinwag, 2017-01-24:
This patch changes the license to what is included in CANopenNode

[0001-change-license.zip](https://github.com/robincornelius/libedssharp/files/726835/0001-change-license.zip)

robincornelius, 2017-01-31:
patch applied as a760324de96a9a92adbab535e591e158cb49e722 

many thanks

closed 2017-01-31 by robincornelius