robincornelius/libedssharp #48: Device Info saving

martinwag, 2017-01-25:
Hi,

when saving  info page to .eds/xml, some info is not exported/saved.

Field                                 ok in .eds export        saved in .xml
Product number               yes                            no
Vendor number                yes                            no

Description                      yes                            no
Creation Date/Time         no                              yes (auto filled)
Modification D/T              no                             yes (auto filled)
Modified by                     yes                            no
Baudrates                        yes                            no
Device settings                yes                            no

I've had a go at fixing this, but wasn't too successfull at finding all the neccessary places in the code. I'm more into embedded systems that GUI design :-)

martinwag, 2017-01-25:
I've tried this using GIT 8f9e79617fc6685693e613efc47fffe0df2c75cc

robincornelius, 2017-01-25:
I'll take a look at those fields, They were saving at one point so probably a regression as things were moved and updated.

The git link you pasted is one to an old commit of mine?

martinwag, 2017-01-25:
Right now, it's the most recent commit.

commit e51e4f59 (2017-01-31): Add in Extra Device Info Fields to xml Fixes #48

Add in the extra fields to the xml, make the CharacteristicsList actualy
a list of Characteristics and use it to store the various extra device
info properties we have. This with the previous commit should complete
the fix.
closed 2017-01-31 by robincornelius