robincornelius/libedssharp #50: Exeption if the "name" string of object or subindex contains the characters "[" and "]"

trojanobelix, 2017-01-31:
An “item with the same key has already been added”
exeption is thrown on  **eds[sectionname].Add(_key_, _value_);** (line 1061)  if the "name" string of the object or the subindex contains the characters "[" and "]" 
like in 
"Filtering time t1 **[s]**"
where the "[s]" is the unit of the "filtering time" in seconds

The "s" is erroneously interpretded and added as a _key_ and so the next string  that contains "[s]" e.g. 
"Filtering time t2 **[s]**" is interpreted as the second occurence of the _key_ "s"

CANchkEDS has no problem wit the "[" and "]" characters in EDS files and checks it as valid.

Might not be needed to change but should be documented

robincornelius, 2017-01-31:
Thanks, my parsing of the section headers is a bit crude, i'll take a look and improve what is going on here

trojanobelix, 2017-01-31:
Wow, very fast. thank you....