robincornelius/libedssharp #51: Subindex written decimal instead of heyadecimal

trojanobelix, 2017-01-31:
The tenth subindex is written as 10 instead of 0x0A and therefore the "10" it is interpreted as 0x10 if checked with CANesdCheck. This results in an error because the number of subindexes (sub0) does not fit the highest subindex

eg: 
# **should:** 
[6401sub9]
ParameterName=Input
ObjectType=0x7
DataType=0x0003
AccessType=ro
DefaultValue=0
PDOMapping=1
;StorageLocation=RAM

**[6401subA]**
ParameterName=Input
ObjectType=0x7
DataType=0x0003
AccessType=ro
DefaultValue=0
PDOMapping=1
;StorageLocation=RAM

# **is:**
[6401sub9]
ParameterName=Input
ObjectType=0x7
;StorageLocation=RAM
DataType=0x0003
AccessType=ro
DefaultValue=0
PDOMapping=1

[**6401sub10**]
ParameterName=Input
ObjectType=0x7
;StorageLocation=RAM
DataType=0x0003
AccessType=ro
DefaultValue=0
PDOMapping=1

robincornelius, 2017-01-31:
Looks like a duplicate of  #44 . This is fixed in git but the not in the last v0.4-beta release.

There are a number of small issues being worked through then i'll do a 0.5 release

trojanobelix, 2017-02-01:
Ah. Yes it is. Sorry, I just tested the 0.4 beta.

closed 2017-02-02 by trojanobelix