robincornelius/libedssharp #54: Date/time in DS306 format for eds file (h:mmtt MM-dd-yyyy) should be culture inrevariant

trojanobelix, 2017-02-01:
Modify/Creation date and timee in DS306 format for eds file (h:mmtt MM-dd-yyyy) should be culture inrevariant.
E.g. german time would be 10:12 in eds instead of 10:12AM, that is not according DS306  (characters in format „hh:mm(AM|PM)“

Added  CultureInfo.InvariantCulture for Creation/Modification date and time.
Unable to push to this repro so her are the changes at line 1230 eds.cs

            fi.CreationDate = fi.CreationDateTime.ToString("MM-dd-yyyy", CultureInfo.InvariantCulture);
            fi.CreationTime = fi.CreationDateTime.ToString("h:mmtt", CultureInfo.InvariantCulture);

            fi.ModificationDate = fi.ModificationDateTime.ToString("MM-dd-yyyy", CultureInfo.InvariantCulture);
            fi.ModificationTime = fi.ModificationDateTime.ToString("h:mmtt", CultureInfo.InvariantCulture);

robincornelius, 2017-02-01:
Thanks for the fix.

If you fork the repo in github, then push to your own copy you can then hit pull request and it gets sent to me and i can pull the changes direct in.

trojanobelix, 2017-02-01:
Ah ok. Thanks, I will try it.

robincornelius, 2017-02-01:
Fixed by commit #55

closed 2017-02-01 by robincornelius