robincornelius/libedssharp #56: RPDO mapping DefaultValue: Wrong length

trojanobelix, 2017-02-01:
When mapping a data type UNSIGNED16 (DataType=0x0006)  the object length in the RPDO mapping parameter doesn't match data type of mapped object.

[2315]
ParameterName=Actual value reference channel
ObjectType=0x8
;StorageLocation=RAM
SubNumber=0x2

[2315sub0]
ParameterName=Number of actual values
ObjectType=0x7
;StorageLocation=RAM
DataType=0x0005
AccessType=rw
DefaultValue=1
PDOMapping=1

[2315sub1]
ParameterName=Actual value reference channel
ObjectType=0x7
;StorageLocation=RAM
**DataType=0x0006**
AccessType=rw
DefaultValue=0
PDOMapping=1

(...)

[1602sub2]
ParameterName=mapped object 2
ObjectType=0x7
;StorageLocation=RAM
DataType=0x0007
AccessType=rw
DefaultValue=0x231501**08**  <= should be 0x231501**16** for UNSIGNED16 
PDOMapping=0

robincornelius, 2017-02-01:
Hmm odd, not sure whats going on here. 

What does it show in the editor for 0x1602 sub 0x02 ? is that the correct value or the wrong one. Also what is shown on the PDO tabs? 

I am wondering if you updated a value, then there is a synchronising issue and the pdos were not regenerated? Did you make any other changes before exporting such as changing 0x2315/01 to a U16 from a U8?

trojanobelix, 2017-02-01:
It is a synchronising issue:
Start: 0x2314/01 and 0x2315/01 = U16, mapped via 0x1602 on Byte 0/1 (0x2314) and  Byte 2/3 (0x2315) 

If you have an PDO mapping and then the data type changes e.g. 0x2314/01 to a U8 from a U16 then the "Assigned PDO mapping list"  in tab view "RX PDO Mapping"is **directly updated** and the mapping shows on Byte 0 (0x2314) and  Byte 1 (0x2315).  Ok so far

But under tab view "Object Dictionary" the default values of 0x1602 still 0x231401**10**.

robincornelius, 2017-02-01:
Ok then that is an easy enough to fix. I will force a PDO update after any changes are saved to any OD entry. This will ensure it can never be out of sync with current data

trojanobelix, 2017-02-02:
Fine.Thanks!

robincornelius, 2017-04-11:
Going to close this now, think the work around is as good as we can currently get

closed 2017-04-11 by robincornelius