robincornelius/libedssharp #57: RX RPDO mapping tab view: dummy mapping of two 16 bit variables fails

trojanobelix, 2017-02-02:
If byte 0 of COB-ID 0x200 is set as a dummy mapping of an dummy int16, byte 1 is automatically set to"-".
Ok so far.

But if byte 2  (next free) is also set to dummy int16 the table gets rearranged in this (faulty) way:

Byte 0 becomes a dummy int**8**, **byte 1** gets the first part of the new dummy int16, byte 2 is automatically set to"-".

Or in other words: When mapping the second 16 bit variable, the first is redefined as a 8 bit variable

Hard to describe, but I think easy to reproduce.

closed 2017-02-02 by robincornelius

trojanobelix, 2017-02-02:
> Closed #57<https://github.com/robincornelius/libedssharp/issues/57> via 9205121<https://github.com/robincornelius/libedssharp/commit/9205121e07b0e5f70176a44ac77523c54d8fbbf9>.
>
>
> technotrans AG ° Firmensitz Sassenberg
> HRB 9086 ° Amtsgericht Münster
> Vorsitzender des Aufsichtsrates: Heinz Harling
> Vorstand: Henry Brickenkamp (Sprecher),
> Dirk Engel, Dr. Christof Soest

Thanks, but I am unable to build, missing icon file:

Schweregrad

Code

Beschreibung

Projekt

Datei

Zeile

Unterdrückungszustand

Error

CS7064

Error open symbol file "(…)\EDS-Editior\Source\libedssharp\EDSTest\Index_8287_16x.ico": The file "(…)\EDS-Editior\Source\libedssharp\EDSTest\Index_8287_16x.ico" was not found

EDSEditorGUI

(..)EDS-Editior\Source\libedssharp\EDSTest\CSC

1

Aktiv

maybe not added?

robincornelius, 2017-02-02:
Sorry it was not added. Done now

commit 9205121e (2017-02-02): Incorrect targets when generating dummy mappings, Fixes #57