robincornelius/libedssharp #59: compiling in Linux/Mono

martinwag, 2017-02-07:
This is not really a problem with libedssharp as it is intended for Windows .net, however I would like to use the editor on Linux OS. So maybe there is a mono guru around :-)

I've tried to compile the editor using mono xbuild / MonoDevelop and ran into the problem that in ReportView.cs the mshtml library is not available.
After disabling the following lines, the compile works:
`//using mshtml;

 //           mshtml.HTMLDocument CurrentDocument = (HTMLDocument)webBrowser1.Document.DomDocument;
 //           mshtml.IHTMLStyleSheet styleSheet = CurrentDocument.createStyleSheet("", 0);
 //           styleSheet.cssText = text;`

Maybe I've just missed the google result that says "install package xyz"?

robincornelius, 2017-02-07:
It might need to be set up very slightly differently for mono,

I'm only using the webbrowser for a very simple formatted report so we could for the moment disable that part. I could still export the reports as html.

however looking at http://www.mono-project.com/docs/gui/winforms/webbrowser/ their should be enough of a web browser for it to work.

Other alternative is https://github.com/cefsharp/CefSharp but thats one hell of a dependency to drag in for simple reporting. Have to see what other alternatives there are.

robincornelius, 2017-02-09:
I've had a play a bit further and found the following 1) the prebuilt exe runs but does nothing and/or crashes on the report page. Building on mono does indeed fail with a missing mshtml dependency. If you grab that from a windows machine and edit the project file to reference it then it builds. But this results in message on the console "that libgluezilla is not found, To have webbrowser support you need libgluezilla installed"... So fine whats libgluezilla. apt-cache search libgluezilla ... turns out its recommended by libmono-webbrowser4.0-cil but does not exist anywhere! The source for it is https://github.com/mono/gluezilla but its eww 9 years old and references xul-runner1.8 ;-( looking bad. You can get the old xulrunner code from the scripts in the gluezilla repository but the automake/conf stuff is so out of date its not playing and refusing to compile a c++ during the ./configure. and no combination of autoreconf/automake/libtoolize etc including -i and -f options is fixing it.

I think the easiest solution is to find a way to disable it for mono and instead invoke a system webbrowser to load my generated html

martinwag, 2017-02-10:
For me, it works when disabling the lines mentioned in the opening post. I don't need an option to run a web browser from the app. When I open the report options, it just displays an empty window (and crashes when I try to save, print...).

robincornelius, 2017-02-10:
The answer is fairly obvious, use a text box in rich text mode as rich text supports plenty of formatting including tables and this will work cross platform.

s-fuchs, 2017-02-21:
Ok, it compiles :-)
Most things work, adding SDO is fine. But the dropdown menu for creating a PDO mapping does not show.. maybe too specific for mono?
A linux version would be very nice!

martinwag, 2017-03-06:
I also have 4.6.2 (debian testing), the dropdown menus don't work on my system too. 
I have no idea how this works (is supposed to work)... 

I get the following in the terminal when I click on a entry
```
SendMessage (127926860, 0x1038, 0x2, 0x7fff56e46b30)
messages.c: A message was found on the uncommon code path: 11 - Bad formatSendMessage (127926860, 0x1038, 0x2, 0x7fff56e46b30)
messages.c: A message was found on the uncommon code path: 11 - Bad formatSendMessage (127926860, 0x1038, 0x2, 0x7fff56e46b30)
messages.c: A message was found on the uncommon code path: 11 - Bad formatSendMessage (127926860, 0x1038, 0x2, 0x7fff56e46b30)
messages.c: A message was found on the uncommon code path: 11 - Bad formatSendMessage (127926860, 0x1038, 0x2, 0x7fff56e46b30)
messages.c: A message was found on the uncommon code path: 11 - Bad formatSendMessage (127926860, 0x1038, 0x2, 0x7fff56e46b30)
messages.c: A message was found on the uncommon code path: 11 - Bad formatSendMessage (127926860, 0x1038, 0x2, 0x7fff56e46b30)
messages.c: A message was found on the uncommon code path: 11 - Bad formatSendMessage (127926860, 0x1038, 0x2, 0x7fff56e46b30)
messages.c: A message was found on the uncommon code path: 11 - Bad formatSendMessage (127926860, 0x1038, 0x2, 0x7fff56e46b30)
messages.c: A message was found on the uncommon code path: 11 - Bad formatSendMessage (127926860, 0x1038, 0x2, 0x7fff56e46b30)
messages.c: A message was found on the uncommon code path: 11 - Bad format
```

Do you have an idea how to fix this - doing the mapping using the default values in OD editor tab would be very annoying...

martinwag, 2017-03-07:
When I try to debug "listView_TXCOBmap_MouseClick()" function, KDE5 hangs up (no reaction to mouse clicks anymore)...

According to the source of the list box (http://www.codeproject.com/Articles/18111/Reusable-ListView-in-C-with-textbox-and-combobox), this class was developed for .net 1.1. Is there no build-in class for the required functionallity in recent .net versions?

robincornelius, 2017-04-11:
I'll have to come back to this and have another go and see if there is a better control to use. May be something under WPF would be better and TBH the entire GUI using a MVC approach would probably have been better. 

I wonder instead of a dropdown inside a list box if a context menu on a regular listbox could serve the same purpose, worst case a pop up window with the ability to select/configure things could also work.

robincornelius, 2017-05-14:
I've spend some time going through this and found the issue with the list box. That is in the commit just pushed. Itt was doing Win32 API calls to get sub item rects and other nasty stuff. I've changed this to 100% .net code but also had to add an additional refresh hook to get it to actually display. Its now working on my debian box

I've also disabled the webbrowser for mono and instead invoke the "xdg-open" command with the path to the report.html file. In theory if your system is set up correctly xdg-open will open your preferred application for the given file (in this case .html) but on my debian box it was not setup and tried to load i.html into monodevelop

Building on mono will still be a little bit of a pain, i want to see if we can improve this to have a seamless build

commit 51619123 (2017-05-14): Make the CustomListView more mono compatable Fixes #59

Remove the WINAPi calls used to get subitem rects and instead just use
.net API and the approprate subitem bounds. Also trigger a redraw of the
listview as mono seems to not handle the subitem show() call unless the
entire list is refreshed
commit 629720c5 (2017-05-14): Don't use web browser control on mono Fixes #59
closed 2017-05-14 by robincornelius