robincornelius/libedssharp #62: Feature - OD Editor text box size alignment

martinwag, 2017-02-07:
This patch makes OD name and desctiption text box size aligned to current window sizes. 
Changes:
- Elements below the decription are now anchored to bottom instead of top
- name text box is additionally aligned to right
- description text box is additionally aligned to right and bottom.

[0001-Changed-behaviour-of-name-and-description-text-fiel-.zip](https://github.com/robincornelius/libedssharp/files/757732/0001-Changed-behaviour-of-name-and-description-text-fiel-.zip)

robincornelius, 2017-02-07:
Thanks for the continual patches ;-). It might be easier if you fork the repository in github then you can commit back to your own version and just hit "pull request" then i can just hit accept and github takes care of the rest. No need to generate and attach patch files ;-)

robincornelius, 2017-02-07:
applied in 40154aa9be490b661d2c18634e35aabb8f6aef51

closed 2017-02-07 by robincornelius