robincornelius/libedssharp #63: Warning object 0x1003: Arraysize vs max-subindex

trojanobelix, 2017-02-08:
On exporting C files I got a warning:

_Subindex descripency on object 0x1003 arraysize: 8 vs max-subindex: 0_

But at 1003 is the Pre-defined error field where SubIndex 0 contains the Number of errors and **not**  the _max-subindex_ like the warning suggests. 

DS301 says:
 _The object entry at sub-index 00h shall contain the number of actual errors that are recorded in the array starting at sub-index 01h._

CANchkEDS has no problem with it.

robincornelius, 2017-02-08:
Yea 0x1003 is a "special case" that behaves like no other OD entry in the fact that the sub index 0 does not reflect the maximum number of entries but the number of errors. 

The warning is too aggressive and needs to have a special work around to not warn for this OD entry as its always like this.

trojanobelix, 2017-02-08:
Sure. It is more polish then an issue. But it might be helpful to have it on a low priority list.

robincornelius, 2017-02-09:
I'll add the exception to the warning on the next batch of commits

commit 9b2e9eff (2017-02-09): Do not issue subindex warning for OD 0x1003, fixes #63
closed 2017-02-09 by robincornelius