robincornelius/libedssharp #64: OD String escapes - wrong char array size

martinwag, 2017-02-08:
When creating a string variable that contains escapes, array size is calculated including the "\" escape char.
Example:
Object 0x100a contains the manufacturer version as string. I would like to default this string to "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0". (empty char array with 36 elements) 
This is correctly placed in CO_OD.c as "/*100a*/ {'\0', '\0', '\0', '\0', '\0', '\0', '\0', '\0', '\0', '\0', '\0', '\0', '\0', '\0', '\0', '\0', '\0', '\0', '\0', '\0', '\0', '\0', '\0', '\0', '\0', '\0', '\0', '\0', '\0', '\0', '\0', '\0', '\0', '\0', '\0', '\0', '\0'},".
However, in CO_OD.h, the array size is twice of what is required: "/*100a      */ VISIBLE_STRING  manufacturerSoftwareVersion[74];".

I've tracked this down to eds.cs file, line 863 (return defaultvalue.Length;), but I have no idea how to tell C# to evaluate the string as ansi C chars with the escapes evaluated...

robincornelius, 2017-02-09:
I think changing defaultvalue.Length to  return defaultvalue.Unescape().Length; should do the trick. I've got some special extension methods loaded for strings that you can see in CanOpenNodeExporter.cs around line 1239 for handing escape characters as they are a little bit of a pain in C#

robincornelius, 2017-02-09:
Fixed in 750e52e49c4e3bcebacbf4941b1590b6cc2c7afc

closed 2017-02-09 by robincornelius