robincornelius/libedssharp #69: Config as NMT master

trojanobelix, 2017-02-21:
Hi robincornelius,

the CanopenNode uses the optional object 0x1F80 for indication of the device is a NMT master.
Suspicious. Normally it should be the boot up that indicates NMT master or not. 

But ok, if we accept: Shouldn't the first bit of this object set when using profile DSP302?

Best regards from sunny Germany

robincornelius, 2017-02-23:
Hi,

0x1f80 is defined in DSP302 as part of the NMT master services. CanOpenNode just uses bit 2 to determine if it should enter operational or stay in pre-operational. The rest of the bits and the rest of the NMT master support are not currently provided within CanOpenNode.

The default NMT master profile provided in the profiles/ folder just contains the required Object Dictionary entries that are set out in DSP302 but as CanOpenNode has no support for this it probably makes sense to default bit 0 to 0. If for some reason you have written the required support in CanOpenNode or some other implementation you can simply flip the bit 0 to 1, but I also suspect you will need to adjust many of the other bits and settings of the extra entries added is DSP302 to match what you support correctly.

trojanobelix, 2017-02-24:
Hmmm, never take a deeper look at NMT  Master implementation of CANOpenNode, but as Janez stated in [this ](https://sourceforge.net/p/canopennode/discussion/387151/thread/45b58fac/?limit=25#a763) thread

>  NMT master is **simple and is actually implemented**. See CANopen.c file and follow "CO_NO_NMT_MASTER".

 it sounds to me that it is fully implemented. 

I can not get the actual DSP302 V4x cause it is not free. Do you have a source for me? PM?

robincornelius, 2017-02-24:
I don't have the v4 of DSP302 only v3.2.1 from :- http://overpof.free.fr/schneider/CAN%20&%20CANopen/CANopen/%A9CiA%20CANCANopen%20CD%20V5.1/html/spec3.htm

CO_NO_NMT_MASTER just enables the CO_sendNMTcommand() function which allows the start,stop,preop and reset etc NMT commands to be sent by the bus. The CanOpenNode stack itself does not call CO_sendNMTcommand() you have to do this from your code. I usually implement a boot delay on all the nodes then on one that i call the master i use this function to start the bus.

That is the minimum needed for DSP302 but this standard also defines a lot more under CanOpen managers etc. One of the managers jobs is not only to start the bus but to boot slave devices in specific ways if needed and to also verify they have booted. None of this additional functionality is currently in CanOpenNode. I did start looking at adding some but have not had the time to really give it a good look.

trojanobelix, 2017-02-24:
It is the first time I had to implement NMT Master and SDO client in my project on an PIC24. In oteher projects I'm the SDO Server and a controller with an IXXAT stack is the NMT master. I'm a little bit lost. Janez examples for that does not get me the devices working. I think because I used a SDO sever 

Do you have an addition example for NMT master and/or   SDO client usage?

I just need to read/write some CAN objects on the SDO server and some basic NMT functionality.

robincornelius, 2017-02-24:
Here is my start up sequence on the master

```
void programAsync(int timer1msDiff)
{
     switch(main_state)
     {
         case APP_STATE_BOOT:
              if(boottimer >= OD_timings.can_rst_comms_delay) 
             {
                 boottimer = 0;
                 CO_sendNMTcommand(CO,CO_NMT_RESET_COMMUNICATION,0);
                 main_state = APP_STATE_WAIT_RST_COMMS;
             }
             break;
            
        case APP_STATE_WAIT_RST_COMMS:         
            if(boottimer >= OD_timings.can_start_me_delay) 
            {
                boottimer = 0;
                CO_sendNMTcommand(CO,CO_NMT_ENTER_OPERATIONAL, CO->NMT->nodeId);
                main_state = APP_STATE_WAIT_START_ME;
            }
            break;
            
        case APP_STATE_WAIT_START_ME:
          
            if(boottimer >= OD_timings.can_start_slaves_delay)
            {
                boottimer = 0;
                main_state = APP_STATE_WAIT_START_NODES;
            }
            break;
            
        case APP_STATE_WAIT_START_NODES:
            if(boottimer >= OD_timings.can_app_run_delay)
            {
                CO_sendNMTcommand(CO,CO_NMT_ENTER_OPERATIONAL, 0);
                main_state = APP_STATE_RUN;
                //enable guards here to prevent errors, if you have guarding enabled
               //and you do a ,CO_NMT_RESET_COMMUNICATION you will get an error message
            }
            break;
            
        case APP_STATE_RUN:                      
            doworkhere();
            break;
     }
}
```

That is called from the canopennode async callback. And clearly i have created an OD entry called OD_timings that contains my delays. 

I also strongly advise that somewhere in the code you ensure the PICs CAN device is set up, then you have a delay before starting the can open node stack. If you don't its a race condition to which device boots first and if no other CAN devices are up on the network the boot message will not receive a bus ACK and you will get a TX PASSIVE error.

trojanobelix, 2017-02-24:
Wow, thanks a lot.
I have seen that  "canopennode async callback" stuff before. Is there a change in the software structure? All example I have seen for the PIC33/PIC just uses a big app main that does all the stuff. But I find it much more clearer to have callback functions like programStart(), programAsync, etc

robincornelius, 2017-02-24:
For the async stuff see https://github.com/CANopenNode/CANopenNode/blob/master/stack/PIC32/main_PIC32.c

and

https://github.com/CANopenNode/CANopenNode/blob/master/example/application.c

I base my PIC32 and PIC24 code off the main_PIC32.c example, just make the appropriate changes to the register setup at the start.

trojanobelix, 2017-03-09:
Thanks a lot. The code is working wel!

My CANopen experiences on embedded side are limited mainly on the DS 301 and the SDO server implemetation.  NMT master function and SDO client functions are done on higher level controls with an IXXAT stack. Do you have experiences with NMT things like network scan and handling of ODs of different nodes on NMT master side? I just do not have an idea if and where this is defined by the CiA. Does CANopenNode support different ODs (=different nodes)? 

Excuse silly questions.

s-fuchs, 2017-03-09:
For the NMT network scan: take a look at [emtas CANopen DeviceExplorer](https://www.emtas.de/en/produkte/canopen-device-explorer) (they have a demo). You can observe the generated CAN messages and maybe figure stuff out ;-)

But i don't think CANopenNode supports different ODs..

robincornelius, 2017-03-09:
As far as I am aware, there are a couple of possibilities for NMT master,

 1) is like my example above, I would not go any more basic than that because if you do you will have boot sync issues with PDOs and i have also discovered devices in the wild that do not correctly start up if you don't issue the communication reset. So always reset then start the bus as per my example

2) Is an extension of the above and is defined in DSP302 where the master basically has a list of slaves and starts slaves in a particular order. It also tracks the state of the slaves from the boot messages.  Also even more extensions define various SDOs that need to be checked before booting such as Device Manufacture and Device version to ensure correct devices are present.. Basically this is a big list on the master and you just walk through it doing SDO Reads to check the data.

In order for this to work the master has to know in advance the list of slaves and what to do with them or what to verify on them. If you don't want to do this just boot the entire network with the simple NMT boot as in 1) and then use guarding to check slaves are alive.

Is that what you mean by network scan? 

What do you mean by different ODs? are you referring to configuration of slave ODs? If so this is "technically" not the job of the master but of the "network configuration" which may or may not be the same node as the master (but often is). Again this is just a list of nodeid/index/subindex/new data to be written to slaves and you can just walk through it updating what you need. I've done things like this by creating an array and a slightly higher level SDOMaster implementation and just pumping each change out as soon as the previous one has updated.

If you want to scan a whole network to see what is connected when you don't know, this is somewhat harder. The easiest way is to at least have an idea of what is there and hopefully have EDS files for each node. My own scanning tool does this (and looks much like what the emats tool does) I load a EDS for each slave then i know what OD entries are present so i can then read/write them and also show human readable info for what the entry is and know what data type to send etc. But if you want to scan an unknown network you have to be more agressive. Use NMT reset messages to restart nodes and look for boot messages then you will have to scan the ODs via SDO to attempt to work out what is there, but this will never be 100%.

trojanobelix, 2017-03-09:
I know this scan from offset printing machines. There could be dozens of different (but known) nodes, depending on the printing machine configuration an the installed peripheral devices. The NMT master do the scan on every (possible) node and awaits an answer. As far as I remember it sends a reset node. If he got a boot up node, master starts scanning the (known) SDOs. So there must be a "known device database" in the background.

In my weird imagination the easy way to build this  "known device database" is to import all EDSes from the possible nodes e.g. with the "device type" as an index

Looks for me as this is a standard problem in network management. So I wanted to know if there is a CiA mechanism how to do it, I will read the DSP302. Thanks for the tip.

robincornelius, 2017-03-09:
Ah i understand what you mean and yes this is part of the DSP302. If you look at v3.2.1 which is linked in an earlier message, you are doing what is in Figure 2 at the very start "Start Boot Slave Process" Figure 3 shows this in more detail. These are called the mandatory slaves which must be present. You can also have other slaves that you don't specifically start just issue a global start for.

You send each slave a reset communication then a start node which is just the following commands

First send this to each slave :-

000 2 aa 82  - where aa is the slave address, this can be done with
CO_sendNMTcommand(CO, CO_NMT_RESET_COMMUNICATION, nodeID);
then the same with
000 2 aa 01  - where aa is the slave address, this can be done with
CO_sendNMTcommand(CO, CO_NMT_ENTER_OPERATIONAL, nodeID);

At the same time as doing that you need to listen on the bus for the NMT replies

Note you will only see the BOOT message from the CO_NMT_RESET_COMMUNICATION because enter operational will not send any additional NMT replies unless you are also using heatbeats

The problems start here though because currently CanOpenNode has no way to receive the boot messages. You will have to intercept them in CO_HBconsumer.c
 as this currently sets up filters for each heartbeat it is configured as a consumer for. So you will need to change this behaviour to listen to ALL heartbeats (change mask and filter) and only set the filter once. In the callback you will need to software filter and one feed the existing heartbeat consumer system and 2 log all the boot states to a big array.

This is something i was going to have a go at implementing, but i've not had the time.

trojanobelix, 2017-03-09:
Thanks a lot for the hints and explanations. DS301 is well defined, but it seems that the other layer parts are not descried in that detail.

trojanobelix, 2017-03-14:
Hi Robincornelius
I got some problems to rearrange the tasked CANopenSocket implementation in a non blocking version.
CANopenSocket and other examples run mainly the CanOpenNode code that contains a lot of blocking While()-loops. But I need to run the main  and just call the Can_app cyclic. 

The CANopenSocket uses a taskmanager and I assume the main code will run on other task so it is also non blocking from main task view

I messed up the SDOclient transfer and do not get where the problem is. I get mystic CAN errors eg  "unknown NMT command..." 
Might be a timing problem
 
Do you have an example for non blocking/non tasked SDOclient transfer ?

robincornelius, 2017-03-14:
Hi,

To do non blocking SDO master you just need the following steps. First do this once to set up a new transfer:-

    CO_SDOclient_setup(CO->SDOclient,0,0,slavenode);
    CO_SDOclientDownloadInitiate(CO->SDOclient,slaveindex,slavesubindex,cantxbuf,datalen,false);

to set up (this is a download example, upload uses CO_SDOclientUploadInitiate ), cantxbuf is the pointer to the databuffer length indicated by datalen; Keep an eye on how big your buffersize is as it really should be SDO_MAX_SIZE or what ever that define is in case you use domain transfers but for regular transfers just pass a buffer of the size you require the data for.

then every cycle call 

    int ret = CO_SDOclientDownload(CO->SDOclient,time1msdiff,1000,&abortcodebuffer);

    if(ret==0)
    {
    //finished
    }

    if(ret<0)
    {
    //ERROR better check abortcodebuffer for more info
    }
    
    if(ret>0)
    {
    // its still doing work take no action and check next cycle.
    }

That is basically it. If you look at he header files for those 3 functions it gives a little more info on the parameters

trojanobelix, 2017-03-17:
Thanx again for your help. 

I found my mistake:
I messed up the time1msdiff things using CANopenSocket implementation in a non blocking version. So the first SDO always timed out.

robincornelius, 2017-04-11:
Think we can close this one as well, feel free to reopen if the editor needs any further changes

closed 2017-04-11 by robincornelius