robincornelius/libedssharp #70: Problem loading a Standard EDS File from Lenze.

mgiaco, 2017-04-04:
Hello Today I tried to load this LENZE_i550_de.eds file which I have attached below. I am not sure what´s wrong but I could not open it. 
Which Version of EDS is supported?

many thanks
cheers
mathias

File:

```
[FileInfo]
CreatedBy=Lenze i-Series Toolchain
Description=CAN eds file by Lenze for the i-Series device
CreationTime=09:01AM
CreationDate=02-10-2017
ModificationTime=09:01AM
ModificationDate=02-10-2017
ModifiedBy=Lenze i-Series Toolchain
FileName=LENZE_i550_de.eds
FileVersion=1
FileRevision=1
EDSVersion=4.0

[DeviceInfo]
VendorName=Lenze
VendorNumber=0x3b
ProductName=i550 CAN standard IO
ProductNumber=0x69353530
RevisionNumber=262145
BaudRate_10=0
BaudRate_20=1
BaudRate_50=1
BaudRate_125=1
BaudRate_250=1
BaudRate_500=1
BaudRate_800=1
BaudRate_1000=1
SimpleBootUpMaster=0
SimpleBootUpSlave=1
Granularity=8
DynamicChannelsSupported=0
CompactPDO=0
GroupMessaging=0
NrOfRXPDO=3
NrOfTXPDO=3
LSS_Supported=1

[DummyUsage]
Dummy0001=0
Dummy0002=0
Dummy0003=0
Dummy0004=0
Dummy0005=0
Dummy0006=0
Dummy0007=0

[Comments]
Lines=0  

[MandatoryObjects]
SupportedObjects=3
1= 0x1000
2= 0x1001
3= 0x1018

[1000]
ParameterName=Device type
ObjectType=0x7
DataType=0x0007
AccessType=ro
[... 9695 lines cut ...]
[404Esub0]
ParameterName=highest sub-index supported (generated)
ObjectType=0x07
DataType=0x0005
AccessType=ro
DefaultValue=0x2
PDOMapping=0

[404Esub1]
ParameterName=Minimalsollwert
ObjectType=0x07
DataType=0x0003
AccessType=rw
DefaultValue=-30000
PDOMapping=0

[404Esub2]
ParameterName=Maximalsollwert
ObjectType=0x07
DataType=0x0003
AccessType=rw
DefaultValue=30000
PDOMapping=0

```

robincornelius, 2017-04-04:
Thanks for attaching the eds file. I found the problem(s)

First problem is the lines like "1= 0x1003" note the space after the = I have modified the parser to ignore whitespace here.

I found one other problem with the import. Object 0x2002 "Gerätemodul" Uses subindex 0 for the highest subindex supported but does not fill in sub indexs 1,2 or 3 so the first actual entry is then 4. This is perfectly allowable in eds format but i was not supporting it correctly.

I have pushed a fix to git but will also do a release as it is in need of an update

closed 2017-04-04 by robincornelius

mgiaco, 2017-04-05:
Wow, many thanks. I would be great if you can make a release for that because I have no running environment.

robincornelius, 2017-04-05:
Already done should be fixed in 0.5.2 release please if you find any other issues let me know but I may be offline for a few days.

commit e03b8b8b (2017-04-04): Discard white space when parsing key/value pairs from eds file and also don't assume first index is key[1] fixes #70