robincornelius/libedssharp #71: Exception in view "RX PDO Mapping"

gbmont, 2017-04-05:
Hello Robin

Thank you for providing the eds editor!

Please try:
open an eds file
switch to register tab RX PDO Mapping
double click on first line unter COB in section "Assigned PDO mapping".. In my case the program raises s an execption.

Friendly regards
Gerhard

robincornelius, 2017-04-05:
I'll take a look as soon as I can, offline for s few days though.

robincornelius, 2017-04-09:
I can reproduce this with Lenze eds file but not the others i have here.  There are two things going on here

I have some bad assumptions that break if subindexes are not continuous. I have fixed many of these but there was another one in the PDO communication parameters that broke in this case. In this eds file the subindexes 0,1,2 and 5 are used for object 0x140X. The other issue that may have triggered this is that i've never seen Event timer used for an RX PDO communication parameters, for TX PDO yes many times but for RX PDO i am not 100% sure this is even valid. However it has uncovered a definite bug in my program that I will push to git tonight and issue a release sometime soon.

commit f1e1372e (2017-04-09): Do not assume array indexes for accessing subobjects, the array ordinal may not equal the sub index number, Fixes #71
closed 2017-04-09 by robincornelius