robincornelius/libedssharp #72: Can't load eds-file

gbmont, 2017-04-06:
Hello

Attached an eds filke from canopennode example. In my opinion, the file looks fine but I can not load it with the the eds-editor

Best regards
Geri

eds file io.eds

; CANopen Electronic Data Sheet
; File was automatically generated by CANopenNode Object Dictionary Editor
; See http://canopennode.sourceforge.net/

[FileInfo]
FileName=IO Example
FileVersion=-
FileRevision=0
EDSVersion=4.0
Description=Open Source CANopen implementation
CreationTime=17:24:43
CreationDate=2016-03-25
CreatedBy=JP

[DeviceInfo]
VendorName=CANopenNode
VendorNumber=0
ProductName=CANopenNode
ProductNumber=0
RevisionNumber=0
OrderCode=0
BaudRate_10=1
BaudRate_20=1
BaudRate_50=1
BaudRate_125=1
BaudRate_250=1
BaudRate_500=1
BaudRate_800=1
BaudRate_1000=1
SimpleBootUpMaster=0
SimpleBootUpSlave=1
Granularity=8
DynamicChannelsSupported=0
GroupMessaging=0
NrOfRXPDO=4
NrOfTXPDO=4
LSS_Supported=0

[Comments]
Lines=5
Line1=EDS File for CANopen device
Line2=Open Source CANopen implementation
Line3=Stack Version: V3.00
Line4=Generated by CANopenNode Object Dictionary Editor
line5=http://canopennode.sourceforge.net/

[DummyUsage]
Dummy0001=0
Dummy0002=1
Dummy0003=1
Dummy0004=1
Dummy0005=1
Dummy0006=1
Dummy0007=1

[MandatoryObjects]
SupportedObjects=3
1=0x1000
2=0x1001
3=0x1018

[... 2704 lines cut ...]
DefaultValue=0

[6411sub6]
ParameterName=Output
ObjectType=7
DataType=0x0003
AccessType=rww
PDOMapping=1
DefaultValue=0

[6411sub7]
ParameterName=Output
ObjectType=7
DataType=0x0003
AccessType=rww
PDOMapping=1
DefaultValue=0

[6411sub8]
ParameterName=Output
ObjectType=7
DataType=0x0003
AccessType=rww
PDOMapping=1
DefaultValue=0

robincornelius, 2017-04-09:
Hi. I'm back and looking at this now.

I've just tried to load the above eds and it loads fine, there are two warnings during load that appear in a separate warning window as follows :-

* Warning parsing FileVersion tried to fit - into Byte
* Unable to parse DateTime 17:24:43 2016-03-25 for CreationTime, not in DS306 format

Both these warnings are valid as FileVersion must be a byte and - is not a valid number and the Date and Time is really strict in its format and this is not 100% correct with the EDS standard. Its possible other eds editors are more relaxed here. But the file is loading and giving me all the object dictionary.

What is going wrong? what error or messages are you getting?

gbmont, 2017-04-09:
Hello Robin

Ok, thank you. You are right. The file is loaded fine with just the two warnings. I thought, there are problems if the warning dialog appears:) Sorry, in this case the file is loaded fine:)

Best regards
Geri

closed 2017-04-09 by robincornelius