robincornelius/libedssharp #73: xml- device  file

gbmont, 2017-04-06:
Hi Robin

Me again:)

I downloaded the free CANopen monitor from  kickdrive.de.
http://kickdrive.de/en/index.html?products_kickdrive_zero.htm

Then I tried to export an eds-file to an xml-file an import it to the program but it seems that subindex are not read correctly.
Furthermore there are xdd-files included in kickdrive zero. I also tried to open it in the eds-editor but I got some errors. 
Maybe it is not ab problem of the eds-editor but maybe you can have a look at it.

It would also be fine if the export an import file open dialog includes the filters *.* *.xdc and *.xdd

Best regards
Geri

robincornelius, 2017-04-09:
I think the subindexes may be related to the #71 I believe this is the same basic problem and I will have a look at this as well.  A quick test does not reveal an obvious problem with the eds files I currently have. Feel free to email over any eds/xml files you are having this issue with. Just send them directly to me at robin.cornelius@gmail.com and i'll see what has gone wrong.

Currently XDD files cannot be loaded, My xml code was based on the format from CanOpenNode's original javascript based eds editor as everything i was doing was related to the opensource stack canopennode. However it looks like XDD is a good standard and the "official" xml file format to replace text eds files. So it should be added to the todo list but i'm not sure when i'll be able to get to this as its a fair chunk of work.

gbmont, 2017-04-09:
Hello Robin

Thank you very much for your fast response and offer! 

In my case, I already had some projectes with CANOPEN and every time, I got an eds text file from the vendors. The free kickdrive CANOPEN-Monitor ist the first program for me that requested the xml-file version. 

Best regards
Geri

closed 2017-04-09 by gbmont