robincornelius/libedssharp #74: File Formats and Problems with other Tools

mgiaco, 2017-04-09:
Hello,

Today I tried some more stuff from this great tool but I had some problems with the XML outputs. For my Canopennode Demo I use this tool. [KickDriveZero](http://kickdrive.de/de/index.html?products.htm
) for some tests. 

This tool uses the *.xdc and *.xdd formats. So for me the first problem was how to get that.
So why not use the standard extension here? So I think the "Save Device XML" should be *.xdc ==> right? But when I try to use that with KickdriveZero it doesn't work correct. Also the Network export doesn't work.

So I tried the free CANeds Editor from Vector => Install CANOe as Demo and then under Tools you will find CANeds. Load the IO.eds file and save the IO.xdc and IO.xdd. Till now for me it is not clear which file is for which cenario. So but the IO.xdc file works well on KickdriveZero. Also the *.xdd file works the same way.

Are you interested in correct your tool to generate the correct files? I can share the files with you but I do not know how because I think it isn´t a good way to fork and add a pull here. So the easier way is use the IO.eds from the CANOpenNode Example and try it with your tool and with the CANeds tool from Vector.

s-fuchs, 2017-04-10:
Hi mathias,
i think you can not generate *.xdd or *.xdc files with this editor (or the old CANopenNode editor).

"Save Device XML" generates a XML in CANopenNode format. 
*.eds describes the device as "eletronic datasheet" (format is *.ini-style).
*.dcf is the device configuration file (same format as *.eds).

*.xdd is the device description (in CANopen XML format, see CiA311).
*.xdc is device configuration, generated from *.xdd with a specific configuration for the device (also in xml format i guess).

robincornelius, 2017-04-10:
I think supporting xdd is a good idea and something that should be added, Looking at the xdd format there is a lot in common with the canopennode xml format and although i suspect direct compatability may prove difficult adding another fileformat in the same way should not pose to much problem.

The way to do this which I will have a look at as soon as i have time is to feed the xdd specification into a C# generator ( Xml2CSharp) which will produce a set of classes which resemble the CanOpenXML.cs file. Add to that a read and write function then it just needs an import/export glue function in bridge,cs and it should work.

I'm still going to keep the existing xml format as it is( should be) compatible with the old canopennode dictionary editor tool but it is worth a look to see how compatible things are as it may turn out that the existing format is simply a subset

robincornelius, 2017-04-10:
I've started an XDD branch on github. Currently i've fleshed out some basic loading logic for XDD files but it chokes when XmlSerializer parses the XDD due to some namespace issues that i don't yet understand. 

I need to have a good look at the two xmls as there are a lot of collisions with what look like identical functions so a bit of tidyup and subclassing could roll both versions into one. Just need to fix XML parsing first...

mgiaco, 2017-04-10:
@s-fuchs okay i understand. @robincornelius That sounds really great.

robincornelius, 2017-04-10:
The XDD branch https://github.com/robincornelius/libedssharp/tree/XDD is now opening xdd files and populating the object dictionary. Lots of other stuff not linked together but doing something useful

robincornelius, 2017-04-11:
I've pushed an alpha test release that supports loading of XDD, please give it a try and see how much it breaks

https://github.com/robincornelius/libedssharp/releases/tag/0.6-xdd-alpha

martinwag, 2017-09-13:
I've had a little play with it. Loading CANopenNode configs from the master branch works. But when I try to save xdd format, the export function crashes somewhere in CanOpenXDD.cs line 221 loop function.

Are there some features (or dead entries from the original CANopenNode editor) that make the export crash?

I've added the file here:
[project.xml.txt](https://github.com/robincornelius/libedssharp/files/1299976/project.xml.txt)

robincornelius, 2017-09-13:
Seems to be an issue with disabled OD entries, fixing that stops the crash
but something else is wrong. I'll investigate further

martinwag, 2017-09-14:
I can confirm that. After exporting (without disabled entries), most of the entries in the OD have different indexes. Sub-indexes seem to be fine.

martinwag, 2017-12-11:
Did you have a look at that?

robincornelius, 2017-12-11:
Should be fixed on branch, the byte order of the 16 bit indexes was
reversed during the conversion to/from xml.

I think XDD mostly works but there are still a couple of rough edges, the
Device Name/Description seems to have two potential locations in XDD and i
need to understand that better. Also the refactoring and extra eds work has
unlocked a bunch of other small bugs (most of which are separate entries on
this tracker).

I'm working through them slowly. Once i get back to a semi stable position,
i'll fork master off as old master and merge xdd to master and try and push
on down this route as its got a lot more work in it now and handles a few
special eds cases a lot better

robincornelius, 2019-02-02:
Closing this, XDD is now the main file format of the app, it still supports save/load of eds/dcf/xml formats but going forward i'm sticking to XDD may be with a few custom attributes where we need to extend things but its pretty much a done thing. nearly all the features are fully supported and eds<->xdd conversion seems to be working fine (i'm sure there are edge cases but for all normal things its fine)

closed 2019-02-02 by robincornelius