robincornelius/libedssharp #75: Save eds, xml, device menu option problems

martinwag, 2017-04-12:
Two small issues with this:
- When exporting EDS, closing and re-opening current project, the location for the EDS file is lost. This means that one has to always re-enter the EDS file path after opening a project. This works as expected with the .c/.h files.
- there is no check in the .c/.h if the files still exists -> after moving a project, exporting leads to a crash.

robincornelius, 2017-04-12:
On point #2 it should not matter that the c/h file exists, i expect what happened was it attempted to export to a folder that did not exist anymore?

robincornelius, 2017-04-12:
499319b836608a6d81051678d309cf6fe4c16241 should fix your issues

martinwag, 2017-04-13:
#1 fixed, thanks!
#2 doesn't crash anymore, but doesn't give an error either. It seems to do nothing? Would using relative paths be a solution for this?

robincornelius, 2017-05-14:
Closing as my first fix and your PR should have taken care of everything here

closed 2017-05-14 by robincornelius