robincornelius/libedssharp #77: EDS data placent to wrong memory

on4ip, 2017-04-17:
Hi. Today I create oun EDS file and find some bug.
If I choose ROM for OD field and save EDS file In Note PAd i see that ;StorageLocation=ROM.
But after Loading EDS to editor i see that field located in RAM. 
If i save XML and load it again filed in Editor the field placement are correct.
It look like some errors during parsing from EDS file.

on4ip, 2017-04-17:
In EDITOR
[image]
IN EDS
[image]

robincornelius, 2017-04-17:
Hi i'll take a look at that.

The EDS file does not really support the StorageLocation it is added in as a comment in the eds (hence the line starting with a ; ) but this still should export the correct data. No other program can support reading this field from eds as its not part of the standard.

The XML format is native to this program and stores all the info we have.

robincornelius, 2017-04-17:
https://github.com/robincornelius/libedssharp/commit/d4b4bcd873b4c0540eafa99370715f28e33b5a0e should fix it. its currently on the experimental branch but ill test it when i have time and merge to master

robincornelius, 2017-05-31:
Merged with master

closed 2017-05-31 by robincornelius