robincornelius/libedssharp #78: Possible issue when generating CO_OD.c

advisoft, 2017-05-12:
If I take the master of https://github.com/CANopenNode/CANopenSocket as at today, and open their _project.xml and then regenerate the CO_OD.h and CO_OD.c files, there is a slight difference in the way the braces are generated (see cursor position):
Original:
[image]

Newly generated:
[image]

It also, doesn't compile, and gives a heap of errors and warnings due to the large size of the file. Here are some:

objDict/CO_OD.c:90:53: note: (near initialization for ‘CO_OD_RAM.LastWord’)
objDict/CO_OD.c:90:56: warning: excess elements in scalar initializer
 /*2406*/ {0x6L, 0x0000L, 0x0000L, 0x0000L, 0x0000L, 0, 0x0000L},
                                                        ^
objDict/CO_OD.c:90:56: note: (near initialization for ‘CO_OD_RAM.LastWord’)
objDict/CO_OD.c:90:10: warning: missing braces around initializer [-Wmissing-braces]
 /*2406*/ {0x6L, 0x0000L, 0x0000L, 0x0000L, 0x0000L, 0, 0x0000L},
          ^
objDict/CO_OD.c:90:10: note: (near initialization for ‘CO_OD_RAM.LastWord’)
objDict/CO_OD.c:91:10: error: extra brace group at end of initializer
 /*2407*/ {0x6L, 0x0000L, 0x0000L, 0x0000L, 0x0000L, 0, 0x0000L},
          ^
objDict/CO_OD.c:91:10: note: (near initialization for ‘CO_OD_RAM’)
objDict/CO_OD.c:91:10: warning: excess elements in struct initializer
objDict/CO_OD.c:91:10: note: (near initialization for ‘CO_OD_RAM’)
objDict/CO_OD.c:92:10: error: extra brace group at end of initializer
 /*2408*/ {0x6L, 0x0000L, 0x0000L, 0x0000L, 0x0000L, 0, 0x0000L},
          ^
objDict/CO_OD.c:92:10: note: (near initialization for ‘CO_OD_RAM’)
objDict/CO_OD.c:92:10: warning: excess elements in struct initializer
objDict/CO_OD.c:92:10: note: (near initialization for ‘CO_OD_RAM’)
objDict/CO_OD.c:93:10: error: extra brace group at end of initializer
 /*2409*/ {0x6L, 0x0000L, 0x0000L, 0x0000L, 0x0000L, 0, 0x0000L},

I guess this could potentially be an issue with their _project.xml file causing this to be generated. Happy to investigate further, I just thought it was worth starting a conversation first.

Thanks

advisoft, 2017-05-12:
I am running the tool on Windows built with Visual Studio 2017. Had issues with various pieces with mono on Linux, but I see there is already an issue for that.

robincornelius, 2017-05-12:
Thanks I'll take a look. If I recall 0x24nn is one of the special groups that need an extra {} at start and end of block. Not quite sure why it's not there. Once you get a brace missing the compile errors are more or less meaningless as the first error is the only important one and that can be some distance from the actual error but your diff shows exactly what's gone missing

robincornelius, 2017-05-12:
I've had a look and it appears the auto array detection for the exporter is only partially working. It appears to be working in the .h files as its generating the array :-

/*2401      */ OD_trace_t      trace[32];

correctly but when initialising this in the .c file it misses off the extra { and } brackets needed to indicate its an array. In the exporter print_h_bylocation() identifies these arrays correctly and does the correct thing for the .h files but this info is not reused for .c and that instead relies on the arrayspecial() function.

I've modified the array detection code that the .h file uses and made it common so the .c exporter can also use it and added its output setting to arrayspecial() so that benefits from this info. And the array detector should now also work better if arrays are non continuous and/or don't start from the first possible OD index
.

advisoft, 2017-05-14:
Great, I have just tested master and the issue is now fixed. Thank you for the very quick response.

closed 2017-05-14 by advisoft