robincornelius/libedssharp #79: compiling in Linux/Mono #2

martinwag, 2017-05-15:
Hi,

Commit 629720c5d28125a59bb5f8c9ed2d196265c1daba breaks mono compile.

adding compilation guard in Form1.cs like in NetworkPDOreport.cs makes the compile work, but I have no idea if that breaks somethinge else...
```
#if !__MonoCS__
            NetworkPDOreport npr = new NetworkPDOreport();
            npr.gennetpdodoc(temp, network);
#endif
```

robincornelius, 2017-05-15:
What error is it throwing? It was working at one point last night on mono,m I was building on debian using msbuild and all was good. I suspect something didn't get committed or pushed.

Guarding NetworkPDOreport will prevent the report from being generated, its only report view that is the real issue as that uses the mshtml compoment. When it was working on my debian box the reports were generating then opening in my chosen default browser.

martinwag, 2017-05-15:
It throws a message that NetworkPDOreport is unknown "Error CS0246: The type or namespace name `NetworkPDOreport' could not be found. Are you missing an assembly reference? (CS0246) (EDSEditorGUI)"
Which is because that class is exculded from build (#if __MonoCS__ in NetworkPDOreport.cs).

robincornelius, 2017-05-15:
Well this is interesting,  Just tried a clean git checkout on Debian.

i put the compile guard  #if __MonoCS__ in the wrong file... however on my debian box i'm using msbuild and i assume Roslyn for the compiler and it just works. I can only think that using msbuild/roslyn does NOT set __MonoCS__  but the runtime  return Type.GetType("Mono.Runtime") is correctly set.

To achieve this i just used debian-testing and "apt-get install mono-complete msbuild" i suspect xbuild produces different results. 

I'll remove the guard completely as its doing nothing

commit 0bfdea15 (2017-05-15): Remove useless compile guard, fixes #79
closed 2017-05-15 by robincornelius