robincornelius/libedssharp #81: <feature> network management / OD values view and modification

martinwag, 2017-05-16:
Hi,

Do you have plans to add network management functionallity? Something like those tools: [emtas](https://www.emtas.de/en/produkte/canopen-device-explorer), [ixxat](https://www.ixxat.com/products/products-industrial/tools/configuration-tools/canopen-device-manager).

The backend side on linux should be quite simple - start canopend, connect to socket, read/write string commands. On Windows, this will be quite a mess as canopend is linux only and porting it would require a driver layer for each can adaptor manufacturer :-(.

I haven't done any further thinking about how to implement network management into the GUI. Currently there is one tab per device, which is used to edit the device. This would require one tab per instance of those devices.

robincornelius, 2017-05-16:
Yes but not sure of the time frame.

I wanted to first bring in the ability to store the extra DCF values in both the text format and the xml format which will then add to the back end the ability to store current/actual values along with min/max, denotation and all those features which make setting up a device easy.

I've already got a C# can monitor app which will monitor an entire bus, log and decode all packets (a bit like wireshark) but also it supports plugins so for a specific job you can have specific decoders so you can have useful decode for PDOs etc. As part of this i've got a full SDO editor that will load the eds/xml files using libedssharp and then allows you to change them on the device in real time. I need the DCF ability to be able to store these changes to actual live nodes.

As for supporting drivers. the easiest way is to implement a driver interface that matches the canfestival API, this already supports about 20 devices on linux and windows and pretty much covers all the common devices. I've already got canopend on windows talking to a USBCAN module and also via nanomsg so i can run a simulated bus of multiple canopen nodes all on windows on the same pc.

One final thing, i'm not 100% sure the network management should be the same app as the eds editor? sure it can share common code like libedssharp but i'm still unsure of best way to proceed.

trojanobelix, 2017-05-17:
I think it should be a different app, because of the different purpose e.g. IXXAT "... tool aimed at  device test, diagnostics, and field service tasks. "
Just my 2 cent

martinwag, 2017-05-18:
Wireshark already has a CAN/CANopen dissector. It can be used with socketcan interfaces.

I agree that separating device editing from network management is a good thing. On the other hand, a first step towards a management tool will mostly have the same functionallity as the Editor - because it is mostly the same, except for changing values in a running instance instead of header files.

robincornelius, 2017-05-18:
That's also the problem with wireshark, it works with socketcan only. Worked great on my raspberry pi but was no use at all on my windows systems where my can interfaces were USB.

I'm going to look at my libcanopensimple library and see if I can publish this. I'm using it as a simple way to connect to an interface e.g. A USB device. Then it also lets you inject any can packet (with appropriate helper api) and it has callback hooks for all packet types as well as supporting sdo client functions. It does not act as a device or do any object dictionary stuff but has been very useful with my logger app and I also use this for sdo configuration of devices. 

I did look at using canopensocket for this but went a different route in the end

robincornelius, 2017-05-19:
@martinwag i've just pushed one of my tools (with a big cleanup) to https://github.com/robincornelius/libCanOpenSimple

It also uses CanFestival as a submodule to load its drivers to get hardware access. I've added an extra driver for the CanUSB device i have here and tested with that and its working nicely so far. The propose of this is to just provide a low level access to a CanBus, then a higher level tool, eg a configurator can talk to nodes via SDO and write out all the required config settings to each node. The libcanopen simple fully supports SDO client transfer (it is NOT an SDO server or infact a CanOpen device at all)

robincornelius, 2017-06-03:
Another small update that is relevant to this issue. The xdd branch now also supports DCF files, these are also in the DSP306 specification and look almost like EDS with a few extras, but contain things like actual/current value and concrete node ids etc so very useful for saving the required configure state.

robincornelius, 2017-09-12:
xdd now also supports current value as well as other device commissioning features, at this point it can save a dcf file that could be loaded by a "CanOpenManager" and used to configure nodes.

martinwag, 2017-12-11:
We're now using UV Software COPbrowser tool for that purpose. This tool can load an eds and with that read and modify values on the network.
For what we do, this is sufficient. For more complex tasks, I agree with you that this should be separated from the ODE.

closed 2017-12-11 by martinwag

robincornelius, 2017-12-11:
Ok no worries.

I've ended up with my own tool that does the same thing. It uses my
libcanopensimple to access the bus, and i've also imported libedssharp into
it to load eds files so it just queries a device based on the eds you load
and allows you to edit the values. One thing worth finishing here in the
eds editor is the dcf file as that is a standard and i suspect your
COPbrowser tool can save them and really my other tool should do the same.
Plus i have a need/want to be able to save actual configured state of nodes
along side the original eds when recording a deploy history. And it makes
sense to do this using the existing dcf standard