robincornelius/libedssharp #82: OD 0x1003 Pre-defined error field - CANopenNode export

martinwag, 2017-05-16:
Hi,

When exporting to canopennode OD.c, inside the array "CO_OD_entry_t CO_OD" Entry 0x1003 max. subindex is filled with value "0". 
In the case of CANopenNode, the creation of the ID 0x1003 answer is handled inside the stack itself. But it needs the max. subindex filled with the actual array size. When value "0" is present, CANopenNode just replies an error "ERROR: 0x06090011 - Sub index does not exist".

When I manually change this value to the actual array size, Entry 0x1003 works as expected. This is also what CANopenNode editor exports.

This value is filled in in CanOpenNodeExporter.cs, line 816. There is already an special handling for 0x1003, but there seems to be no variable/method in od object to get actual available subindexes instead of max subindex.

Thanks!

martinwag, 2017-05-16:
Can we just move line 816 "nosubindexs = od.getmaxsubindex();" into the if statement above it? There seems to be no reason to overwrite it, except that if() is true?

trojanobelix, 2017-05-17:
0x1003 is a "special case" that behaves like no other OD entry in the fact that the sub index 0 does not reflect the maximum number of entries but the number of errors.
0x1003 is the Pre-defined error field where SubIndex 0 contains the Number of errors and **not** the max-subindex like on all other objects

**DS301 says:**
> _The object entry at sub-index 00h shall contain the number of actual errors that are recorded in the array starting at sub-index 01h._

So it might be a CANopenNode issue.

martinwag, 2017-05-17:
CANopenNode has a special handler for creating Subindex 0 value when it's read (CO_Emergency.c, CO_ODF_1003() ). The value in the configuration file in this case only indicates the maximum array size to CANopenNode and can not be accessed from the network.

trojanobelix, 2017-05-17:
CO_ODF_1003() reflects _DS301 7.5.2.4 Object 1003h: Pre-defined error field:_

Reading Sub 0 returns NumOfErrors
Writing Sub 0 to 0 resets errors, all other values INVALID_VALUE.

martinwag, 2017-05-18:
I created fix https://github.com/robincornelius/libedssharp/pull/84 to handle this.

Writing sub 1003-0 in CANopenNode actually has a bug that makes it not work, see https://sourceforge.net/p/canopennode/discussion/387151/thread/a9d12d35/

trojanobelix, 2017-05-18:
Doesn't it make sense to create an issue report on [github.com/CANopenNode/CANopenNode/issues?](https://github.com/CANopenNode/CANopenNode/issues)

robincornelius, 2017-05-22:
i see https://github.com/CANopenNode/CANopenNode/pull/38 so hopefully that will be accepted as thats were the fix really needs to go.

Its possible we could do something here but really working around other bug in upstream tools is not maintainable long term and getting the pull accepted in CanOpenNode is the way forward

martinwag, 2017-05-22:
Fix https://github.com/CANopenNode/CANopenNode/pull/38 is a different bug (regarding writing 1003-0 from network). 
CANopenNode still needs to know the actual array size for object 1003.

robincornelius, 2018-01-21:
I've just reopened this issue on xdd branch, when fixing some array/subindex handling, i'll readdress it properly

closed 2018-01-22 by robincornelius