robincornelius/libedssharp #86: EDS export [comment] section

martinwag, 2017-05-22:
The canopen conformance test tool requires an section [comment] like this

```
[Comments]
Lines=1
Line1=dummy to silence conformance tool
```

If that section doesn't exists, testing fails with an error. Should we place a dummy entry or can we use that for something useful?

trojanobelix, 2017-05-22:
on Vectors CANeds it is on the "File  Information" tab. Maybe we could create some similar.

[cid:image001.png@01D2D2F4.5D6FB1E0]

technotrans AG ° Firmensitz Sassenberg
HRB 9086 ° Amtsgericht Münster
Vorsitzender des Aufsichtsrates: Heinz Harling
Vorstand: Henry Brickenkamp (Sprecher),
Dirk Engel, Dr. Christof Soest

robincornelius, 2017-05-22:
DSP306 v1.1 "If not specified otherwise, all sections and entries in this paper are mandatory" so yea [Comments] needs to always be exported. 

It is supported but nothing is populating it and its not generating if there are no comments

robincornelius, 2017-05-22:
And oh is

[Comments]
Lines=0

acceptable to the conformance tool?

martinwag, 2017-05-22:
I don't know if that conforms to DSP306, but yes, it is accepted by the conformance tool :-D

commit ae52894b (2017-05-31): Force comments list<string> to be created always so that the [comment] section is always written out to eds files fixes #86
closed 2017-05-31 by robincornelius