robincornelius/libedssharp #89: Additional "Memory Type"

martinwag, 2017-06-20:
The editor currently supports the memory types "ram", "rom", and "eeprom". Those are also supported by CANopenNode stack and driver template.

I now need another eeprom section to store factory data in my device (e.g. calibration values) which musn't be cleared when restoring via "restore default parameters". From what I've seen of the editor source code, this shouldnt be too difficult (please correct me if I'm wrong).
But I now have multiple choises, of which I don't know which way is better

1. Just replace "EEPROM" entry with "EEPROM1", "EEPROM2", and so on.
- easy to implement
- driver template needs to be changed by user as this only supports one eeprom region. How do we tell that to the user?
2. Add a text field where the user can add memory regions
- I have no idea how to implement that into the gui
- It's more obvious that it's in the users responsibility to handle this region
3. Add memory region "FACTORY_DATA"
- easy to implement
- this solves only **my** problem

What do you think about those options?

robincornelius, 2017-06-20:
2 would be the way to go.

Exporter side :-

On the exported side change StorageLocation to a Dictionary<string> then print_h_bylocation() becomes a foreach(string location in StorageLocation) and the associated comments can be moved in to the print_h_bylocation() function.   export_OD_def_array() the same and getlocation() needs modifying to return name.

StorageLocation should be managed by libedssharp not the GUI so if it reads a file with new locations it can just add them to the current list on a per eds instance basis.

Saving as eds or GUI just saves the new string names in the appropriate locations. in EDS its a comment anyway so harmless. in the XML need to be careful that the parser loads it correctly and adds it to the list if needed, but thats all.

GUI Side :-
2 options
1) Change the list box to a combo box so that sections can be added as required, downside to easy to mess up, so no
2) Add a separate window to allow extra items to be added/show a list of what items are defined, pretty easy and from UX makes obvious sense

robincornelius, 2017-07-17:
Arrg 27 days ago!, where does time go? do you still need this or have you worked around something?

martinwag, 2017-07-18:
I'm going to implement this, but with a very low priority, as right now having basic functionallity is more important than saving production data.

martinwag, 2017-08-01:
implemented by https://github.com/robincornelius/libedssharp/pull/99

closed 2017-08-01 by martinwag