robincornelius/libedssharp #93: Adding editor version info

martinwag, 2017-07-12:
I need the current git version info of the editor available inside the editor and the output files. In C this would be quite simple, just execute a git command in the makefile and then pass this as define to the compiler.

In c# this seems to be quite a pain. I've came as far as this: https://github.com/martinwag/libedssharp/tree/add-version-string . This generates a file version.txt as pre-compile step, adds this file to the project output and reads it on application start.
However I don't really like reading the version number from a file. I also have no idea on how to properly pass this version number to the export functions.
Can you give me a hint :-)

robincornelius, 2017-07-12:
On 12 July 2017 at 15:01, Martin Wagner <notifications@github.com> wrote:

> I need the current git version info of the editor available inside the
> editor and the output files. In C this would be quite simple, just execute
> a git command in the makefile and then pass this as [...]

Best i have here is "msbuild /p:DefineConstants=" but if you use that it
clobbers all defined constants including DEBUG etc so remember to set the,
all here. Would also need to "Exec Task" the git executable from msbuild.

There is also
https://gitversion.readthedocs.io/en/latest/usage/msbuild-task/   .. i've
no idea of implications if this is not installed (eg does build break)? If
this does just work it seems ideal

However I don't really like reading the version number from a file. I also
> have no idea on how to properly pass this version number to the export
> functions.
> Can you give me a hint :-)
>

For the exported CO_OD files the entry point
is CanOpenNodeExporter::export(string folderpath, EDSsharp eds) Add a
string to that parameter list with your version info then it can be just
added to what ever comments in the exporter you would like.

If you want to write to EDS as well as a comment then you can just access
the" eds.c.comments" list and append strings as needed and this will appear
in the comments section

Writing to the XML is a bit more tricky, really the comments in the eds.c
class should just appear there but currently they don't. There are no
defined comments sections in the XML probably because this xml format is
the old CanOpenNode javascript exporter one. the DCF branch uses proper
CanOpen DCF format (as well) so should have places to put comments

To add to the existing XML, either choose a place, eg a List such as
Features or add a new variable in to CanOpenXML.cs in the approprate
XmlRoot, then the magic happens in Bridge.cs this has two functions one
converts eds->xml the other xml-> eds so you would need to add in something
to pull your version in and out here as well.

Hope that helps

Robin

> —
>
>

martinwag, 2017-07-13:
Thanks for your answer. I've got it working by now.
I've misread the part on stackoverflow where it said how to embed the version.txt file into the executable. With that, it's fine to me.
I've added the version string to the .c/.h export functions only, as I don't want them in the user files.

This still has the problem that it needs the "git" command in the environment and the full git repo on the build machine to build the application. The command works both on windows and linux without problems. It creates a string like "0.5.2-32-gcc8eb8a-dirty" while developing. The "dirty" goes away when building on a clean repo, the "-32-gcc8eb8a" commit info goes away when tagging a commit.

If you want to include that feature, I can create a pull request.

robincornelius, 2017-07-13:
On 13 July 2017 at 10:47, Martin Wagner <notifications@github.com> wrote:

> It feature, I can create a pull request.
>
>
>
It sounds useful, I've certainly done things like this before in C++
projects. As long as it works gracefully if git is not found then please
submit a PR

Thanks

Robin

martinwag, 2017-07-14:
That's the problem I'm having. If git is not available, the build fails. VS just shows a message that the command failed with error code 9009.

> Fehler		Der Befehl "git describe --tags --long --dirty > "E:\workspace\libedssharp_\EDSTest\version.txt"" wurde mit dem Code 9009 beendet.	EDSEditorGUI		
	
Obviously, VS is missing a "show message in english" option :-(

martinwag, 2017-07-14:
Fixed that by putting "|| exit 0" at the end of the command. This is actually supported by windows cmd :-)

robincornelius, 2017-07-17:
Its not an issue for this repository (as the release system auto tags) but if you try this on a repo with no tags yet the git command will fail, to fix this add --always

closed 2017-07-17 by robincornelius