robincornelius/libedssharp #97: Unhandled exception when exporting CanOpenNode c/h files

ChrisAlphabet, 2017-07-25:
Hi,

I am trying to generate the `CO_OD.c` and `CO_OD.h` needed for messing around with CanOpenNode-Socket but the Object Dictionary Editor is showing an error message when I try to generate the files.   

> Unhandled exception has occurred in your application. If you click Continue ....
> 
> ... Index was out of range. Must be non-negative and less than the size of the collection. 
> Parameter name: startIndex

When I clicked continue I saw that the files were generated but they seem incorrect.   
For example in `CO_OD.h`, `DRV_DISTO` is defined twice, at `3445` and at `36dc`.

The EDS I used can be found [here](http://www.kollmorgen.com/en-us/products/drives/servo/akd/) under the software/firmware tab and I used the release 0.6-XDD-alpha.

Just to give you an idea of what I am planning to do is to create a virtual canopen device that I can use for testing while I wait for hardware. I had hoped to change the EDS used in CanOpenNode-Socket example to test my understanding of CANOpen using the EDS of the drive I will eventually be controlling. Any suggestions for getting something like that up and running would be much appreciated!

Thanks,
Chris

robincornelius, 2017-07-25:
Hi,

I've tried the EDS in the xdd branch and the newer version (not yet released) does not produce the exception you discovered.

The duplicate entries in the CO_OD.h file are still present however and i can see why this happens. If you reuse the same name in the "Name" section of the object that is used to generate the code for the .c/.h file.  The fix here is for me to track names already used and append 1.2.3 etc if needed to make them unique. I need to track this in a number of places across the exporter so i'm not going to rush a fix as it will probably be wrong.

If you want to get working on this quickly, grab visual studio 2015/2017 community (free) then grab my code from the xdd branch and load that in to visual studio. And run, that will give latest version. Then just rename the DRV_DISTO on object 0x36dc to  DRV_DISTO1 etc and hit export then repeat for any other duplicates you may find.

ChrisAlphabet, 2017-07-31:
Thanks! Worked a treat :+1:

closed 2017-07-31 by ChrisAlphabet