robincornelius/libedssharp #102: PDO mapping less than 8 bits

JuTakata, 2017-09-07:
Hello,

I would like to map some bits from different sub indexes and it seems that this is not possible.

Object 3700 : 
Sub index 01 : VarA (2 bits)
Sub index 02 : VarB (3 bits)
Sub index 03 : VarC (3 bits)

And I would like to achieve : 
PDO1 Byte 0 Bit 0 = VarA (bit 0)
PDO1 Byte 0 Bit 1 = VarA (bit 1)
PDO1 Byte 0 Bit 2 = VarB (bit 0)
PDO1 Byte 0 Bit 3 = VarB (bit 1)
PDO1 Byte 0 Bit 4 = VarB (bit 2)
PDO1 Byte 0 Bit 5 = VarC (bit 0)
PDO1 Byte 0 Bit 6 = VarC (bit 1)
PDO1 Byte 0 Bit 7 = VarC (bit 2)

I loaded my existing EDS file from one other EDS editor (CANOpenNode html) which was configured like that and I get this error for each mapped sub indexes : 
[image]

The PDO is mapped by one byte for each subindex : 
[image]

Please is there a way to achieve this ? Or this is not supported yet ?

Cheers
Julien

s-fuchs, 2017-09-07:
Hi Julien,
CANopenNode does not support PDO mapping on the bit level (neither the editor).
The mapping function only allows per byte mapping: https://github.com/CANopenNode/CANopenNode/blob/master/stack/CO_PDO.c#L228

regards s-fuchs

robincornelius, 2017-09-07:
Hi,

No the editor does not yet support this. The warning appears as its failed
to understand the currently set mapping so that is the correct thing to do
if the mapping is not understood.

I was not actually aware that you can do this with CanOpenNode as the PDO
mapping parameters in canopen node only allow one to set
index/subindex/datasize to be mapped. However it should be on the feature
list for the editor as other implementation do support this and it should
be able to read/write these eds files correctly.

Regards

Robin

JuTakata, 2017-09-07:
@s-fuchs  Thank you for your answer you have right, I did not tried this yet I was modifiying my OD but I was requiring Domain variable so I just switched to Libedssharp before testing this.

@robincornelius Thank you for your answer, as @s-fuchs answered this is not supported by CanOpenNode, I thought it was and I will probably have to add this feature because this is in my customer specification.

Regards. Julien

martinwag, 2017-09-08:
Usefullness depends on your requirement. Think about mapping a 401 output to a 401 input. With bit mapping you can assign any of the inputs to any output you want. With byte mapping, you only can assign groups of 8.
CANopenNode currently stores the mapping in an array, so making this bit-mappable means increasing array size from 8 to 64.

JuTakata, 2017-09-08:
Hi

@martinwag I will modify my code to be byte mapping yes a bit-mapped approach would increase array size and I already have some memory issues.

I would like to add TIME_OF_DAY type to your EDS Editor but I am not very familiar with C#.
Should I open a new issue and describe the extra code required for .c and .h and someone could add this feature later ? 

Regards

Julien

robincornelius, 2017-09-13:
Please yes, open a new issue for TIME_OF_DAY

closed 2017-09-13 by JuTakata