robincornelius/libedssharp #104: TIME_OF_DAY data type

JuTakata, 2017-09-22:
Hi,

I would like to suggest a new enhancement to support TIME_OF_DAY feature :
 
On CO_OD.h there should be the time of day typedef : 
`typedef timeOfDay_t  TIME_OF_DAY;`

And for CO_OD.c there should be brackets for each initializer : 
```/*2020*/ {0xeL, {0}, {0},```

I also added timeOfDay_t declaration to my CO_driver.c : 
```   
typedef union {
     unsigned long long ullValue;
     struct {
           unsigned long ms:28;
           unsigned reserved:4;
           unsigned days:16;
           unsigned reserved2:16;
     };
}timeOfDay_t;
```

I will suggest this to CANopenNode but I did not tested this yet.

Best regards

Julien

kingd123, 2017-12-07:
JuTakata's request and suggested implementation, gets my vote too :-)

TIME_OF_DAY Data type is called for the CiA402 power drive profile, for Object 6406 'Motor calibration date'

It's the only CiA402 object that requires TIME_OF_DAY Data type mind !

JuTakata, 2017-12-07:
Thank you for your support =)

closed 2018-01-21 by robincornelius