robincornelius/libedssharp #105: Is CO_OD.c/CO_OD.h a LGPL?

xoduddk123, 2017-10-11:
Hello.

I've been using the object dictionaries editor with firefox.
But I learned from martinwag that there is something called libedssharp. ([Where do I download the Object Dictionary Editor?](https://github.com/CANopenNode/CANopenNode/issues/62))
It is very convenient because it can be used as a GUI without using firefox.

I have downloaded the edseditor-Setup-v0-6-xdd-alpha-81-gb562769.exe file from https://github.com/robincornelius/libedssharp/releases/tag/0.6-xdd-2.
I exported CO_OD.c, CO_OD.h with the installed Object Dictionary Editor (v0.6-xdd-alpha-81-gb562769).
Below is the top part of CO_OD.c, CO_OD.h.

https://github.com/robincornelius/libedssharp/blob/5c8f76f75f49c63db1757c9a359dc50c61b9184b/libEDSsharp/CanOpenNodeExporter.cs#L260-L314

The 265 lines are licensed as LGPL.
However, the following 277 lines are declared as GPL v2.
And below is a GPL linking exception (classpath exception) declared.

1. I think the 265line in libedssharp/libEDSsharp/CanOpenNodeExporter.cs should be GPL v2 like https://github.com/CANopenNode/CANopenNode/blob/master/example/CO_OD.c.

2. I think the URL on line 267 should be modified to https://github.com/CANopenNode/CANopenNode.

Is there a bug in the alpha version?

I'm sorry I do not have enough English. :)

Thanks!

robincornelius, 2017-10-11:
There has been discussion on CanOpenNode mailing list about this a few times and its a little bit of a mess. The licence text in CanOpenNodeExporter.cs was copied from CanOpenNode and this has been updated a few different times. May be its got an old version, or got copied incorrectly.

But something looks wrong, it does look like we have the words for LGPL and GPL+ClassPath mixed up a bit.

xoduddk123, 2017-10-11:
Will it be changed to GPL+ClassPath later?

I'm sorry I do not have enough English. :)
Thanks!

robincornelius, 2017-12-07:
Just a comment. I don't think it "really" matters what the licence is in CO_OD.c CO_OD.h as they are an output from a tool, which is based on your input data. 

From the GPL licence

"Activities other than copying, distribution and modification are not covered by this License; they are outside its scope. The act of running the Program is not restricted, and the output from the Program is covered only if its contents constitute a work based on the Program (independent of having been made by running the Program)."

The code only comes with a header as the template was an exact copy of the one inside CanOpenNode, and it will be definition be the licence of CanOpenNode that matters here

trojanobelix, 2017-12-07:
In my opinion the output files shall (read: must) not declare a licence type.
Think about: Should e.g. all OpenOffice output files contain this licence contract? 
If you agree, no one could use OO in a commercial environment.

closed 2019-01-30 by robincornelius