robincornelius/libedssharp #106: The v0.6-xdd-alpha-81-gb562769 version can not generate an EDS file.

xoduddk123, 2017-10-11:
Hello.

There is something strange about using the Object Dictionary Editor.
I used https://github.com/robincornelius/libedssharp/releases/tag/0.6-xdd-2 in the release.
I could not create an EDS file in the Object Dictionary Editor using exe file and source code.

I've clicked "Save eds + xml + Export device", but I need to create an EDS file first.
[image]

[image]
But I have not found a way to save the EDS file.

I know that the version is pre-release, so I downloaded and built the source of https://github.com/robincornelius/libedssharp.
I built the source of the master and had an Open EDS, Save EDS button and was able to create an EDS file.[image]

I'm sorry I do not have enough English. :)

Thanks!

robincornelius, 2017-10-11:
Hi,

On XDD branch the eds and other file types are all under Save Device FIle, but now there are multiple types :-

[image]

Then you can select the type of file to save :-

[image]

The XDD branch supports more file types.

the actual xdd files are still a bit broken and i've not had time to look at them again

xoduddk123, 2017-10-11:
Thank you for your kind reply.
I was able to see your answer and create an EDS file in the XDD branch.
[image]

Thank you!

closed 2017-10-11 by xoduddk123