robincornelius/libedssharp #107: Object's index and subindex are strange when creating xdd file.

xoduddk123, 2017-10-12:
I created a Device File with the contents of the xdd branch. (See address)
The xml, eds, and dcf files are generated correctly.
However, when saving a file with xdd, the index of the Object and the number of the subindex are swapped.

The following is a part of the xdd file saved after applying DS301_profile.xml and DSP302-NMTMaster.
[image]

The index of "Device Type" is stored as 0010.
The subindex of "Pre-defined error filed" is stored in the order of "0100", "0200", "0300".
I think the place is swapped.

And when I open the xdd file that I saved, the Object Dictionary Editor will display the following.
[image]

Something seems wrong.

I tested the code by changing the code as shown below and confirmed that the index and subindex of Object are saved normally in xdd.
https://github.com/xoduddk123/libedssharp/commit/4ad7f6e5611bb9b403e1507a9617e8e97fcbb598

I'm sorry I do not have enough English. :)

Thanks!

[device file(xdd_branch).zip](https://github.com/robincornelius/libedssharp/files/1378133/device.file.xdd_branch.zip)

robincornelius, 2017-10-12:
Hi, yes i knew about the byte order problem but didn't get around to pushing a fix.

Can you send a pull request from your branch and i'll include it

xoduddk123, 2017-10-12:
I've sent a request to https://github.com/robincornelius/libedssharp/pull/108.

robincornelius, 2017-10-12:
Fixed by PR #108

closed 2017-10-12 by robincornelius