robincornelius/libedssharp #109: CANopenNode fails to compile due to OD_NMT_Startup.

xoduddk123, 2017-11-10:
Hello.

Using CO_OD.h generated by libedssharp causes an error when compiling CANopen.c of CANopenNode.
[image]
The errorStatusBits error was resolved by recreating CO_OD.h.
However, the OD_NMTStartup error is slightly strange.

The CO_OD.h generated by libedssharp is defined as follows.
[image]

In CANopen.c of CANopenNode, it is supposed to be used as OD_NMTStartup.
https://github.com/CANopenNode/CANopenNode/blob/533801574055c5dacf60fb6b18930c288336dce1/CANopen.c#L647-L654
[image]

So when compiling with CO_OD.h using libedssharp, an error occurs.
I first modified CANopen.c to fix the error.
But I think either libedssharp or CANopenNode need to be modified.

I modified libedssharp as shown below.
https://github.com/xoduddk123/libedssharp/commit/2268581331a0e703a78fec305ef8365f1bba46e5

I'm sorry I do not have enough English. :)

Best regards.
kwon

martinwag, 2017-12-08:
Can this be the same problem as this one: https://github.com/robincornelius/libedssharp/issues/110#issuecomment-348989681 ?

I'm not on the xdd branch, and for me 

```
/*1f80 */
        #define OD_1f80_NMTStartup                                  0x1f80
```
is generated.

robincornelius, 2017-12-11:
No its not the same bug. #110 was a hard coded string with a space in. This looks like a different hard coded string with a space,  its being solved my modifying the default profile file for NMTMaster.

xoduddk123, 2017-12-15:
I confirmed that the source has been modified.
I am happy to contribute to the project.

Best regards.
kwon

commit bd672745 (2017-12-11): NMT Master profile had incorrect name (with space) for object, fixes #109
closed 2017-12-15 by xoduddk123