robincornelius/libedssharp #110: EDS file check fails.

xoduddk123, 2017-11-10:
Hello.

I tried to create and use an EDS file with libedssharp.
I use LS's XBL-CMEA PLC.
[image]
When registering EDS file created by libedssharp in software, registration fails as below.
[image]
"선택한 파일은 CANopen EDS파일이 아닙니다." 
 → "The file selected is not a CANopen EDS file."

I thought the problem was a software bug.
So I installed CANoe 10.0 (64bit) of Vector.
There is an EDS related tab called CANeds in the program.
[image]
I can load and check the EDS file from the CANeds tab.

If [Check] → [Check file] of CANeds is executed, the following error occurs.
[image]

It says that "SupportedObjects" can not be found.
When i open the EDS file, it is "Supported Objects".
[image]
There is one space.

I have tested and the following modification will successfully check the EDS file.
https://github.com/xoduddk123/libedssharp/commit/c8eca9d277d76bd4d75c375db9f6b83e0c631cd6
[image]
[image]
[image]

I'm sorry I do not have enough English. :)

Best regards.
kwon

robincornelius, 2017-12-04:
I can see what has happened. Looks like the XDD branch has a bug and is incorrectly putting a space in Supported Objects

xoduddk123, 2017-12-15:
I confirmed that the source has been modified.
I am happy to contribute to the project.

Best regards.
kwon

commit 2bb04fee (2017-12-07): Refactoring code had introduced a space in the "SupportedObjects" key - Fixes #110
closed 2017-12-15 by xoduddk123