robincornelius/libedssharp #111: CANOPENNODE implementation

thanga2597, 2017-11-20:
how to i use the canopennode example i/o code to develop the own canopen i/o module

robincornelius, 2017-11-20:
You have to implement all your own logic inside the functions in application.c 

There are a number of functions there to fill in. Mostly programAsync() which is where you will do most of your work and program1ms() which you do any time critical work (called every 1ms do not block)

The EDS editor allows you to edit and define the object dictionary and connect up PDOs etc without editing code so then CO_OD.h will have many useful defines you can use in your logic to read/write the Object Dictionary.

thanga2597, 2017-11-20:
Do you have example canopen i/o program ?

robincornelius, 2017-12-07:
There is most of an example in CanOpenNode under CANopenNode\example the only thing you need to add that is specific to your processor is the code in application.c

For example if you wanted to support digital input and output :-

(example for a PIC and i've assumed you are using PORTA and PORTB and bits 0-8 in each port. for a real application you will have to decide what to connect to what based on your electrical schematic

This will run on the 1ms callback and upate the PIC/object dictionary with the current state of the hardware...

void program1ms(void)
{

        //read digital in
        OD_readInput8Bit[0] = PORTA;

       //write digital out
      PORTB = OD_writeOutput8Bit[0];

    OD_readAnalogueInput16Bit[0] = ReadADC(0);
    OD_readAnalogueInput16Bit[1] = ReadADC(1);
    OD_readAnalogueInput16Bit[2] = ReadADC(2);
    OD_readAnalogueInput16Bit[3] = ReadADC(3);

}

Really this is out side the scope of this tool. This project/tool is for generating eds files and the CO_OD.c and CO_OD.h. Its probably best to learn to use CanOpenNode first without this tool so you understand what it does. When you come to needing to edit CO_OD.c/h then this tool will help

closed 2017-12-07 by robincornelius

thanga2597, 2017-12-07:
I don't know how to communicate between PLC and pic microcontroller through
canopen by using the canopen node for making I/o module