robincornelius/libedssharp #113: Writing out wrong key with multiple PDO mappings

Atrer, 2017-12-06:
On both the TX PDO Mapping tab and the RX PDO Mapping tabs, if you add another PDO using the "Add new PDO" button (so that there are 2 total) and then save the EDS file.  Re-opening the file will fail with the following exception:

```
System.ArgumentException: An item with the same key has already been added.
   at System.ThrowHelper.ThrowArgumentException(ExceptionResource resource)
   at System.Collections.Generic.Dictionary`2.Insert(TKey key, TValue value, Boolean add)
   at libEDSsharp.EDSsharp.parseline(String linex)
   at libEDSsharp.EDSsharp.loadfile(String filename)
   at ODEditor.ODEditor_MainForm.openEDSfile(String path, filetype ft)
```

This is caused by the application writing out the wrong key for sub indexes 1601sub0 and 1A01sub0 when saving.

Here is an excerpt from the diff of the eds file after adding that second PDO.  Bold for emphasis:
```
 [1A00]
 ParameterName=TPDO mapping parameter
 ObjectType=0x9
@@ -868,6 +1061,93 @@ AccessType=rw
 DefaultValue=0x00000000
 PDOMapping=0

+[1A01]^M
+ParameterName=TPDO mapping parameter^M
+ObjectType=0x9^M
+;StorageLocation=ROM^M
+SubNumber=0x9^M
+^M
**+[1801sub0]^M**
+ParameterName=Number of mapped objects^M
+ObjectType=0x7^M
+;StorageLocation=RAM^M
+DataType=0x0005^M
+AccessType=ro^M
+DefaultValue=0^M
+PDOMapping=0^M
+^M
+[1A01sub1]^M
+ParameterName=mapped object 1^M
+ObjectType=0x7^M
+;StorageLocation=RAM^M
+DataType=0x0007^M
+AccessType=ro^M
+DefaultValue=0x00000000^M
+PDOMapping=0^M
```

Replacing that 1801sub0 with 1A01sub0, for TX PDO or the 1401sub0 with 1601sub0 for RX PDO by hand fixes the issue.

robincornelius, 2017-12-07:
Should be fixed by commit 3d6173bc6a673bc3b419181c3cfb65be52452d83 on xdd

robincornelius, 2017-12-07:
Test builds available at https://www.dropbox.com/sh/mby6s6dofqkte1e/AABJnzMlkYW1MEzXskmOrxlUa?dl=0

Atrer, 2017-12-07:
Perfect, thanks!

closed 2017-12-07 by Atrer