robincornelius/libedssharp #118: Access all areas

kingd123, 2017-12-07:
For ARRAYs, could the editor allow please, Access to be editable for the subindicies, instead of greying

Why ?

Because the CiA301 standard requires that for some of mandatory objects (1003 1010 1011 1016 1029),
sub-index 0 (ie max sub-index) is const, but sub-index 1 to n are rw

Presently with the editor, for ARRAYs, sub-index 0 to n all follow the Access for the overall object

Consequently, the XML has to be edited instead in eg Notepad, in respect of those objects

The editor is happy though, to display objects so configured, and to generate C/H files, EDS file, etc, OK

So, I guess what I'm asking, is please could the editor not grey ARRAY sub-indicies.
And instead allow them to be individually edited - like for RECs.

robincornelius, 2017-12-07:
There is/was/should be an override for this for the special indexes that use this behaviour.In the top right window on the Object Dictionary. If you (RIGHT) click on subindex 0 a context menu should appear and it should let you change the max sub index value. I believe this does what you require? Let me know

kingd123, 2017-12-07:
Many thanks for the reply on this. I've just checked, and that right-click menu is there, I never knew about before !

The menu indeed includes a 'Change max sub index' option. But I'm afraid, that's a different thing, from the issue I mention.

The menu has a small problem I noticed: The existing Number of elements always seems to show as 0.

I could see the menu being useful though, for correcting the count. Although of course normally, one would right-click on subindicies 1 to n, and chose + Add sub item or - Remove sub item, which of course automatically increments or decrements the count to match.

Btw I like the use of the + and - signs in the UI. It's a helpful visual to speed work ..

robincornelius, 2017-12-07:
Hi,

I will check back on my next round of fixes and look at this one properly.

I've fixed a small number of things and also rigged up a very crude
approximation of a continuous build server. Latest binaries can be found at
:-

https://www.dropbox.com/sh/mby6s6dofqkte1e/AABJnzMlkYW1MEzXskmOrxlUa?dl=0

That link will refresh every time i build a release build. So at any given
time could be in any state! But should provide an easy way to test latest
commits

Regards

Robin

kingd123, 2017-12-07:
Hi,

Ah OK, thanks
Great on the continuous build server, ping me when there's anything you'd like me to try
'So at any given time could be in any state!' :-) that sounds like my software !

Regards,

David

robincornelius, 2017-12-09:
I've banged together a quick fix that may or may not be enough. I've not yet allowed full sub index access as i'm not 100% sure that is necessary. Instead i already had one "special case option" for object 0x1003 but i've expanded that now to include 2 additional cases.

We now have (on the parent access type) 3 special options  0x1003 rw/ro, 0x1010 const/rw and 0x1010 const/ro which do as their name implies set the subindex 0 to the first value and all the other subindexes to the 2nd value which should give the required combinations for the special cases.

If this still does not work i'll have to revisit this system and rewrite it as its a bit fragile currently

kingd123, 2017-12-10:
Hi Robin,

Many thanks for this. I've just tried the new release (v0.6-xdd-2-12-gbfbebf5), and I'd say these 3 special cases look to be a workable solution, and a better one than the ungreying I was suggesting :-)

I must admit, I didn't spot the dropdown '0x1003 rw/ro' special case in the old release (v0.6-xdd-alpha-81-gb562769), but now I look, I can see.

So yes, I'd say this special case solution could be rolled out also to 0x1011 const/rw, 0x1016 const/rw and  0x1029 const/rw, if you agree they are needed. I hand-edited my XML, but would be good if the editor could. 0x1029 const/const can be covered without need for special case, ie just const for all. 

One thing that foxed me, is that after I chose in the Access Type dropdown, 0x1010 const/rw or 0x1010 const/ro, and then Save Changes, the change took effect. But for both, the dropdown then reverted to showing rw, and the XML and EDS contain just rw, and C contains the equivalent attribute value. I think they should be ro for 0x1010 const/ro.

Apart from that, I can understand the dropdown reverting from special case to eg standard rw, as that's what the EDS has to contain, so the reverting is presumably intentional for that reason.

On a different note, I can see from a file delta, that the generated EDS now has the SupportedObjects nospace fix, great :-)

And the H has the TIME_OF_DAY datatype addition, per your commit comments. I've tried that straightaway, for the CiA402 power drive profile Object 6406 'Motor calibration date', all builds AOK :-)

Best regards,

David

robincornelius, 2019-01-30:
I thought i got to the bottom of the issues here, and it was ages ago. So if anything is outstanding and still wrong in latest builds please create a new issue for the specific bits ;-)

closed 2019-01-30 by robincornelius