robincornelius/libedssharp #120: PDO numbers

kingd123, 2017-12-08:
Hi Robin,

This is very much a wishlist item / suggestion for easy of use, rather than an issue at all.

But, on the TX PDO Mapping tab, it would be really nice, if an additional column could be added, on the left, with TPDO number. In other words, just a simple column, headed TPDO, containing static numbers, 1, 2, 3, etc.

Of course, that's implied, by the order of the rows - the first is 1, the second is 2, and so on. And you can of course, suss the TPDO number, from what's in the Slot column

But if you have a lot of TPDOs, as we do (22, with big unused gaps, don't ask!), it's a bit of a pain,
to do the quick head maths, from slot number hex, to TPDO number decimal. Or to count down the rows.

Much nicer, if the editor just told you the TPDO number, nice and clear, in a new column, on the left.

And perhaps too, swap the COB and Slot column order - as the Slot is a fixed number, but the COB is configurable

Similar suggestions apply for the RX PDO Mapping tab.

Best regards,

David

trojanobelix, 2019-10-08:
Works fine for me.

robincornelius, 2026-10-20:
Maintenance review: this original repository is obsolete; ongoing development and fixes are in [CANopenNode/CANopenEditor](https://github.com/CANopenNode/CANopenEditor).

Fixed: sparse PDO list displays actual PDO numbers derived from communication indices.

Implementation and validation: [merged PR #237](https://github.com/CANopenNode/CANopenEditor/pull/237), with [library regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/IssueRegressionTests.cs), [XDD loading tests](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/XddLoaderTests.cs) and [actual WinForms editor regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/EditorRegressionTests/EditorRegressionTests.cs). GitHub CI passed the .NET Framework 4.8.1/.NET 8 Debug/Release matrix. Linux/Mono, Vector CANeds and firmware execution are not claimed by these Windows checks.

Closing this historical report as resolved/implemented in the maintained successor; the obsolete repository is not receiving duplicate implementation backports.

closed 2026-10-20 by robincornelius