robincornelius/libedssharp #122: read-only/const VISIBLE_STRING max size/initializer

iurly, 2018-01-05:
I'm trying to export some read-only strings (i.e. identifier information which needs to be calculated at startup).
I was thinking about defining it in RAM and initializing it at some point during startup. Ideally I would only need to set a pointer to some RAM or FLASH area where it is stored.
The only way I found to achieve this though would be to declare it as a VISIBLE_STRING and provide an initializer which is long enough to contain the maximum allowed string length, and then have the startup code overwrite it.
Is there a better way to do this?
Thanks!

robincornelius, 2018-01-06:
Really this is a question for the CanOpenNode list as i only export the CO_OD file in the format that can open node expects...

That said i'll say what i know, using a direct string there is probably not much you can do as this is the way the OD works. However you may be able to achieve something much nicer by using the OD access override functions   CO_OD_configure()  this function registers a callback that triggers on every OD read/write and you can handle what happens and data storage yourself, so its possible to use a user buffer in ram/rom/whatever and handle it anyway you like.

iurly, 2018-01-06:
Thank you, I also though about the callback way though I was about to leave it as a last resort.

So I started with the constant initializer approach but I found a few issues:
- I found no way in Object Dictionary Editor v0.5.2 to specify a dummy string of, let's say, 12 bytes, with 12 being a readable number. So I would have to use "\0" 12 times or "123456789012".
- The constant string seems to generate the wrong number in the .length field of the CO_OD_entryRecord_t (apparently, it outputs the right size in decimal format, but it mistakenly prefixes it with a 0x). If confirmed, that's an issue with this project.
- Even ignoring the two issues above, I found no easy way to change the "string" length (i.e. the above field) in software. That's because CO_OD_entryRecord_t is declared as const and therefore allocated in read-only flash. I would also have to use the full array name and the numerical subindex which I don't like.

So I guess I'll have to go with the callback approach then.
Thanks!

robincornelius, 2018-01-21:
Issue #2 "The constant string seems to generate the wrong number" is fixed in git and on the dropbox preview builds. 

#1 Is a good addition for the editor, being able to set a dummy length for a string would be useful

robincornelius, 2026-10-20:
Maintenance review: this original repository is obsolete; ongoing development and fixes are in [CANopenNode/CANopenEditor](https://github.com/CANopenNode/CANopenEditor).

Fixed: empty strings and minimum string capacities produce valid storage and matching OD lengths.

Implementation and validation: [merged PR #237](https://github.com/CANopenNode/CANopenEditor/pull/237), with [library regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/IssueRegressionTests.cs), [XDD loading tests](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/Tests/XddLoaderTests.cs) and [actual WinForms editor regressions](https://github.com/CANopenNode/CANopenEditor/blob/171bd0c5f6cc29aa1fa1faf1f9ffcbb4fd662dc1/EditorRegressionTests/EditorRegressionTests.cs). GitHub CI passed the .NET Framework 4.8.1/.NET 8 Debug/Release matrix. Linux/Mono, Vector CANeds and firmware execution are not claimed by these Windows checks.

Closing this historical report as resolved/implemented in the maintained successor; the obsolete repository is not receiving duplicate implementation backports.

closed 2026-10-20 by robincornelius