robincornelius/libedssharp #126: denotation field

iurly, 2018-01-31:
Hi,
[foreword: I don't know much about EDS]

I would like to specify some per-object custom information in an EDS so to be used by a separate program (written e.g. in Python). I was thinking of the Description field but noticed how this is not saved onto the EDS (in .ini format). I assume the .eds format does not allow that.
So I thought maybe the XDD format would contain that, but it looks like that's not the case either.
So I noticed the new v0.6-xdd-alpha-81-gb562769 release supports a "denotation" field but it looks like that's not saved, neither in the .XML nor in the .XDD file. 
Is that a bug? What's the purpose of this "denotation" anyway?

Thanks!

robincornelius, 2018-01-31:
Hi,

Everything you never wanted to know about EDS can be found
in DSP306_V1_0.pdf which if you google should turn up plenty of results.

XDD is the xml version of EDS + some extra bits. but i can't remember at
this point if it also includes denotation (and i've not finished that bit)
or if that is in yet another file format. As there is another file format
that is like xdd (its still xml but a slightly different schema)

What you want is a DCF file, DCF is like an EDS but with extra fields,
including denotation. (Its there on the save/load device file type options)

Theory, EDS is suppose to be a generic description of a device and supplied
to you by a HW vendor. You set some variables and make some changes then
you record those changes as a DCF file which should be (in theory) able to
be used by a field programming tool to configure each device on the canopen
network.

Quoting DS306

"When using a DCF in a concrete application it is useful to assign
application specific names to the objects. This can be done by simply
renaming the entry value of ParameterName. As an alternative it is possible
to write the changed names into an extra entry called Denotation. "

 [6000sub1]
 ParameterName=Dig.Input Lines 0-7
 Denotation=ApplicationSpecificName1 ...

So if you are writing a config tool, you can save friendly names in
Denotation that might be better for end users.

Hope this helps.

Robin

iurly, 2018-01-31:
Hi Robin,

thanks a lot for the thorough explanation. That's much clearer now.
Here's my understanding about OpenEDSEditor. I can envisage two usage roles, vendor and user.
As a device vendor, I would use it to define an EDS for each hardware product. So I would:
- Use the CANOpenNode .XML format as a working format
- Export the .c/.h files to automatically generate the code for CanOpenNode
- Export .EDS and/or .XDD to be given to the users along with the device
- In this role, many fields like denotation, "Actual Value", and the "Device Commissioning" don't really make any sense as they only make sense with a physical instance. All but denotation are marked as such by a reddish background.

As a user, I could use it to commission the physical devices I have (some of which might be identical), so I would:
- Open one instance of the .XML/.XDD/.EDS file for each physical instance I want to install
- Populate the commissioning fields as above, Actual Value and possibly denotation for each device and object I'm interested in
- Save a single .NXML file as a self-contained working format to describe the whole network
- Export one .DCF or .XDC for each device in case I want to use a commissioning tool which only understands those

Is the above understanding correct?
If that's the case, then there's indeed a bug in that the denotation is not saved, neither in the .XML (where it doesn't actually belong) nor in the .NXML file where it belongs. It's correctly saved in the .DCF though. The .XDC does not currently seem supported at all, but it should have that.

Thank you for reading all the way through the end!

robincornelius, 2018-02-01:
Hi

> thanks a lot for the thorough explanation. That's much clearer now.
> Here's my understanding about OpenEDSEditor. I can envisage two usage
> roles, vendor and user.
> As a device vendor, I would use it to define an EDS for each hardware
> product. So I would:
>
>    - Use the CANOpenNode .XML format as a working format
>    - Export the .c/.h files to automatically generate the code for
>    CanOpenNode
>    - Export .EDS and/or .XDD to be given to the users along with the
>    device
>    - In this role, many fields like denotation, "Actual Value", and the
>    "Device Commissioning" don't really make any sense as they only make sense
>    with a physical instance. All but denotation are marked as such by a
>    reddish background.
>
> Yep this is correct.

> As a user, I could use it to commission the physical devices I have (some
> of which might be identical), so I would:
>
>    - Open one instance of the .XML/.XDD/.EDS file for each physical
>    instance I [...]
Yes also correct, however i question if the ODE editor is the best tool for
the job. Part of the overall plan i'm working on involved having
libedssharp as a library which does all the EDS/XML/XDD decoding/encoding
and using that in another tool for exactly this kind of work.

I've got another app (yet to be made public) called CanMonitor which
connects using CANUSB hardware (other devices are in theory supportable and
it should be capable of using canfestival drivers), As its name suggests
its a monitor app, but it has a plugin architecture and one plugin is the
SDO editor. This loads in (using libedssharp) a device file and then lets
you edit in real time any SDO variable on a device. What this should do
(but does not yet) is use things like denotation and then at the end save
the changes as a DCF/XDC or allow one to import a DCF/XDC and apply this to
the device directly as a commissioning role.

> Is the above understanding correct?
> If that's the case, then there's indeed a bug in that the denotation is
> not saved, neither in the .XML (where it doesn't actually belong) nor in
> the .NXML file [...]

The XML/NXML are our own formats so we can do what we like with them and
including denotation etc is not a bad thing to do, i certainly don't want
to invent yet another format.

> Robin
>

iurly, 2018-02-01:
Thumbs up for CanMonitor! Please let me know if you're willing to share it, whenever you have time. Thanks again for your invaluable help!

closed 2019-01-30 by robincornelius