robincornelius/libedssharp #127: Description for sub-objects is not saved

iurly, 2018-01-31:
If you set a description for an object it is correctly saved onto the .XML file and restored.
If you set a description for a sub-object (i.e. sub-index object of a REC or ARRAY object), it is not saved.
Is this a limitation of the XML schema used (I have no idea if this is specified anywhere), or is it just a bug?

robincornelius, 2018-02-01:
Its a limitation of the xml used, it was originally taken from the
CanOpenNode javascript ODE to keep compatability. i'm pretty sure i can add
subfields and keep it compatible. Long term it would be nice to move over
to the official XDD format if possible. But currently there are still
untested areas within that that are not finished.

iurly, 2018-02-01:
I see, but does XDD support a description field anyway? I didn't see it anywhere, not even for the main objects.
Thanks!

kingd123, 2018-02-04:
+1 iurly, I was just about to post the same
Typed a few sub descriptions, and wondered why they'd gone
Workaround for the mo, is to use idx description, and refer to my subs there

robincornelius, 2018-02-26:
I've not got as far as implementing this in the old XML but i've been looking at the XDD and there is provision for description in there and the below is as much a note for my own remembering:-

Each object also has an associated "parameter" and this contains an array called itemsField and one of the allowable types for this is  [System.Xml.Serialization.XmlElementAttribute("description", typeof(vendorTextDescription))]  and this sounds like the missing description!

kingd123, 2018-02-27:
Hi Robin, ah interesting discovery, sound promising. Regards, David

robincornelius, 2019-01-30:
Closing as this was fixed a while back, the description of subobjects  is correctly saving in the xdd file format. Its unsupported in the other formats.