robincornelius/libedssharp #135: Compatibility entry

kingd123, 2018-04-11:
Hi Robin,

Hmm, I think you want to be omitting the 'compatibility entry' element (subindex 4), from TPDO 1800s

Why?

Because the CiA301 standard (301v04020006_cor3.pdf), says so:

' Sub-index 04h is reserved. It shall not be implemented; in this case read or write access leads to
the SDO abort transfer service (abort code: 0609 0011h). '

Which is exactly what the CANopenNode stack does :-)

Eg, if I try reading 0x1800.4, via SDO, I get that abort code.

If you search 301v04020006_cor3.pdf, for 0609 0011, you'll find there's a few 'hits'
But I think with the editor, the TPDO 1800s, are the only ones affected

I've tried removing the 'compatibility entry' from my TPDO 1800s

But unfortunately the editor doesn't handle correctly gaps in subindex numbering,
in terms of the generated files

A workaround, is to strip out all the entries like this, from the generated EDS file
Although that's obviously a pain, to have to do, each time I generate

[1800sub4]
ParameterName=compatibility entry
ObjectType=0x7
;StorageLocation=RAM
DataType=0x0005
AccessType=rw
DefaultValue=0
PDOMapping=0

Best regards,

David King

trojanobelix, 2018-04-12:
The only hit of a search of

> Sub-index 04h is reserved. It shall not be implemented; 

belongs to the RPDOs not TPDOs
in chapter 7.5.2.35 Object 1400h to 15FFh: RPDO communication parameter, page 132!?

Abort code 0609 0011 is just the "Sub-index does not exist." for unimplemented sub indexes
--

robincornelius, 2018-04-12:
I'll take a look. Most of the code to support subindexes with gaps exist,
there are a couple of locations where this might not be true and it just
needs porting over to the correct internal API to support this. I thought i
had done this a while back but i suspect i missed some locations which used
the ordinal method (which is really not great and error prone).

kingd123, 2018-04-12:
Hi Robin,

Apologies for reply here, not on GitHub - I've forgotten my password (again !),

and been locked out for a bit !

Ah, per my original post, you'll need to search for the string, "0609 0011"

Then, you should find, in 301v04020006_cor3.pdf, four hits:

Indeed, hit 2, at bottom of page 131, is in an RPDO section of the PDF.

(Likewise also hit 3, on page 132, although that's for the SYNC subindex).

Hit 4, however, middle of page 139, is in an TPDO section of the PDF.

That section starts way back on page 136, so it's not at all obvious, for sure..

I've attached my copy of the PDF, for convenient reference.

The second attachment shows, for information, the errors that inclusion of 18xx.4 subindicies,

cause in the CANbus monitor tool we use (emtas CANopen Device Explorer). This when we do a

read of the complete dictionary, or just the Communication section.

Best regards,

David

robincornelius, 2019-10-23:
Ok we are stuck with this because of the the internals of canopennode, maybe we can revisit in the future now the stack is active again and clean this up

closed 2019-10-23 by robincornelius