robincornelius/libedssharp #136: \n vs \r\n

gotocoffee1, 2018-04-16:
Hey,
I just wanted to note that e.g. in the CanOpenNodeExporter.cs the line ends are inconsistent. When using string literals with @ characters, \r\n is automatically appended to the end of the line, otherwise only \n is used manually. It would probably be better to use Environment.NewLine instead.
Maybe like this:

```c#
sb.AppendFormat("/*{0:x4}*/ {1}," + Environment.NewLine, od.Index, formatvaluewithdatatype(od.defaultvalue, od.datatype));
```
or the newer syntax
```c#
sb.AppendLine($"/*{od.Index:x4}*/ {formatvaluewithdatatype(od.defaultvalue, od.datatype)},");
```

My compiler always complains about that. If you want I can make a pull request.

robincornelius, 2018-04-16:
Hi,

Yea pull request would be great thanks!

Append line is much nicer.

robincornelius, 2019-01-30:
Thanks for this, i totally forgot about it. Catching up on some outstanding work!

closed 2019-01-30 by robincornelius